bigtreetech/PandaTouch #1: Will no connect to my mesh network...

tamlaras, 2024-02-02:
It show the devices in the mesh as separate networks (all called the same - 21ChurchStreet) and will not connect to any of them.
[image]

Markus5578, 2024-02-03:
Die selben Probleme habe ich auch, habe Probleme mit meinem Mesh Fritz!Box Netzwerk. Entweder er verbindet sich, oder es funktioniert einfach nicht. Das echt ärgerlich.

G4Cab, 2024-02-04:
Do you have special characters in your wifi password?

tamlaras, 2024-02-05:
No, no spesial characters, just alphanumeric.

cpuks, 2024-02-08:
It usually goes automatically after powering on off the screen via button in back - had same couple times when I couldn't connect but after on/off it connected in seconds, another thing that worries me - wifi antenna is really weak - I have to stand next to my unifi router for Panda Touch to connect to printers - cannot do this on printer as signal is so weak although both X1C and P1P + 2 x esp32 (BL LED) have no problems at all with wifi...

looxonline, 2024-02-26:
@tamlaras Were you able to make any headway into this? Are you using WPA3? That could be the cause of the issue. You may need to fall back to WPA2/3 compatibility mode.

looxonline, 2024-02-26:
> > @tamlaras Were you able to make any headway into this? Are you using WPA3? That could be the cause of the issue. You may need to fall back to WPA2/3 compatibility mode.
> 
> Fortunately I have another (weaker) wifi network I can use. I still can't get it working on my mesh. The mesh if configured as WPA2.

OK. I am going to see if we can build a special firmware version for you that spits out diagnostic info via the USB-C connector and helps us understand what is happening with your mesh wifi connection. The problem with mesh networking is that there were no real standards ratified for it before manufacturers decided to start implementing it and some manufacturers did some interesting things to get it to work but those things interfere with some devices.

looxonline, 2024-03-04:
> > > > @tamlaras Were you able to make any headway into this? Are you using WPA3? That could be the cause of the issue. You may need to fall back to WPA2/3 compatibility mode.
> > > 
> > > 
> > > Fortunately I have another (weaker) wifi network I can use. I still can't get it working on my mesh. The mesh if configured as WPA2.
> > 
> > 
> > OK. I am going to see if we can build a special firmware version for you that spits out diagnostic info via the USB-C connector and helps us understand what is happening with your mesh wifi connection. The problem with mesh networking is that there were no real standards ratified for it before manufacturers decided to start implementing it and some manufacturers did some interesting things to get it to work but those things interfere with some devices.
> 
> That would be great, thank you.

OK so I have worked with the firmware engineering team on this and they mentioned that the stock firmware already outputs some debug messages over serial.

To read these messages you would need to connect the panda touch to a PC/mac. It will then enumerate as a COM port. You can see the messages by connecting to the COM port using a serial tool as described here: https://learn.adafruit.com/windows-tools-for-the-electrical-engineer/serial-terminal

If you are using a mac then I suggest using CoolTerm instead but the process is very much the same.

To detect which is the correct COM port on windows it is often best to go into the device manager with the device connected, show the COM ports in the serial device list, disconnect the panda touch and then see which COM port disappears from the list. A mac should have a name written next to the COM ports in the list which makes it obvious which one to select.

After you connect, you'll see debug messages appear. Try to connect to your mesh network and then let us know what the debug messages say. If we need more debug info then we will tweak the firmware and get it to spit more out.

[image]

dilixi, 2024-11-19:
> It shows the devices in the my mesh as separate networks (called the same thing - 21ChurchStreet) and will not connect to any of them. ![Untitled](https://private-user-images.githubusercontent.com/132651539/301949210-a2a29861-3ee1-4d97-87c4-311ac4f4a1e0.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzIwMDUyNTAsIm5iZiI6MTczMjAwNDk1MCwicGF0aCI6Ii8xMzI2NTE1MzkvMzAxOTQ5MjEwLWEyYTI5ODYxLTNlZTEtNGQ5Ny04N2M0LTMxMWFjNGY0YTFlMC5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQxMTE5JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MTExOVQwODI5MTBaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1hZTUzZTcwOTE2MzEwOGExMDkxM2FkOTQ2NjQzODlmOGYwYWI1NDA5NzczNWQ5YmZmNTc3NDU1MzI2NTk4OGRiJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.pYmJ_k5TsOTC6b2Njjdt4PNaOJi2l38VG7SJSK_-ud8)

Hi,
Does it can connect to your mesh network now?

closed 2025-03-22 by dilixi