bigtreetech/PandaTouch #12: Panda graphic

DoTsTeR77, 2024-02-05:
I would like the option to change the Panda graphic to something custom or at least be able to change to the bigtreetech logo.. Its nice but is a little too cartoonish for my taste.. thank you..

UBEXHH, 2024-02-05:
Same! Makes it look cheap. Like a toy.

Qjeezy, 2024-02-09:
I agree with this.

andypiper, 2024-02-12:
My device has a BTT logo by default? Running firmware 1.0.1.

DoTsTeR77, 2024-02-12:
That's there when there is no print job.  When you start a print the panda
screen cones up..

Aerocats, 2024-02-12:
I would like a print model preview in that space as I start the print.

looxonline, 2024-02-26:
We will consider this but it may only be for V1.0.4 release as the next release will be focused on critical bug fixes and critical functionality additions.

looxonline, 2024-02-26:
> I would like a print model preview in that space as I start the print.

Model preview is a challenge because the preview is embedded within the zipped 3mf which means that the 3mf needs to be downloaded locally and unzipped. We may be able to do this if there is an external USB connected but that is still a variable that we need to look into.

DoTsTeR77, 2024-02-26:
Regarding previews.. How about a small program/app running on a PC/smartphone/mac/whatever on the local network that can take care of a few things that the bambu touch does not have the horsepower for..  for example.. send it the .3mf file.. have it spit back a .jpg preview for the bambu touch to use.. it's use could be totally optional.. just an idea to throw in the hat..

DogWzrd, 2024-02-26:
Regarding previews.. could this be done with a mqtt call to the printer like what the Home Assistant integration does ( cover image)? 
Link to their repository:  https://github.com/greghesp/ha-bambulab

looxonline, 2024-02-26:
> Regarding previews.. could this be done with a mqtt call to the printer like what the Home Assistant integration does ( cover image)? Link to their repository: https://github.com/greghesp/ha-bambulab

This is also something that we have considered but it is not getting it from the local MQTT, it is getting it from the cloud. We are entirely able to implement this route and it was actually on the roadmap two months ago but we have paused it because we are waiting to hear more about the bambu API.

T0MANY, 2024-02-27:
Being able to customize the graphics of the home screen would be fantastic, being able to add a user-customized image would be a very welcome feature!

ktdevil6, 2024-02-27:
> We will consider this but it may only be for V1.0.4 release as the next release will be focused on critical bug fixes and critical functionality additions.

I hope so I think this is an important thing to have the graphic of what is printing on the screen.

looxonline, 2024-02-28:
V1.0.3 will have a file that contains all of the PNG graphics and you can edit them to your heart's content.

gpambrozio, 2024-08-09:
@looxonline looking at the code for the home assistant integration, it looks like this is accomplished with a direct connection to the printer. See [here](https://github.com/greghesp/ha-bambulab/blob/main/custom_components/bambu_lab/pybambu/bambu_client.py#L68)

gpambrozio, 2024-08-10:
I managed to reverse engineer a lot of stuff and figured out a way to update the image every time you start a print (if you have a home assistant with [this integration](https://github.com/greghesp/ha-bambulab) installed.

See [here](https://github.com/bigtreetech/PandaTouch/issues/108#issuecomment-2278962552).

I'm probably going to create a proper AppDaemon app to make it easier to install using hacs.

bigtreetech, 2024-08-10:
> @looxonline looking at the code for the home assistant integration, it looks like this is accomplished with a direct connection to the printer. See [here](https://github.com/greghesp/ha-bambulab/blob/main/custom_components/bambu_lab/pybambu/bambu_client.py#L68)

Hey Man! Thanks for the tips. Since the last comms on this thread we have figured out method to get print images to work and the functionality will be rolled out in future firmware releases.

dilixi, 2024-11-19:
> I would like the option to change the Panda graphic to something custom or at least be able to change to the bigtreetech logo.. Its nice but is a little too cartoonish for my taste.. thank you..

Hi,
there is [online tool](https://pandatouch.us.kg/) to change the panda image, and you can use it on local by this [rep](https://github.com/bigtreetech-beta/PandaTouch_Addtion_IMG).

And please close this issue thanks.

closed 2024-11-20 by BTTJUSTIN

BradCOnGitHub, 2025-05-07:
The linked online tool does not exist anymore

dilixi, 2025-05-07:
> The linked online tool does not exist anymore

Hi,
We have updated its link:
https://ptimgtool.bttwiki.com/