bigtreetech/PandaTouch #13: AMS filament change leaves nozzle temp on..

DoTsTeR77, 2024-02-05:
AMS filament change leaves nozzle temp on..  it should go back down to zero after it is done..

UBEXHH, 2024-02-05:
Also the external manual load

It was the same load-temp problems week after week with xtouch. Annoying!
BigTreeTech should quickly update the scripting of the buttons!

looxonline, 2024-02-26:
This is not true. It follows the exact same cool down profile that the stock BBL load function does. It will immediately go down to a lower temperature once the filament is loaded but remain there for several minutes. After that it will drop to an even lower temperature and remain there for a while. After that it will turn off completely.

closed 2024-02-26 by looxonline