bigtreetech/PandaTouch #7: Z-Axks+

UBEXHH, 2024-02-05:


DoTsTeR77, 2024-02-05:
I confirm this.. I do not like the up and down being reversed.. Please change that or allow us to Invert it..

UBEXHH, 2024-02-05:
After some thoughts, I *think* they mean the "z-room" wider or narrow, according to the arrows.
I have never seen it this way and it is really confusing.

Logically it should be:

1- Up 10+
2- Up 1+
---
3- Down 1+
4- Down 10+

Highlander-bdj, 2024-02-12:
I agree, this needs to be flipped.

Vantage-DS, 2024-02-13:
> The controls for the Z-axis are reversed on display. Lowest button > up Highest button > down
> 
> Should be reversed!
> 
> Why is there an up AND down arrow on all 4 buttons?

I guess it's related to raising Z height, like a Bed slinger.
Up raises the Z height, but for this to happen the bed goes down.

No idea why the icons are like they are though 😄

wilk-polarny, 2024-02-18:
This is terrible, counter-intuitive and the opposite of how the original UI of the built-in screen and the Bambu Studio UI works. Had the heatbed crashed into my nozzle. Now the heatbed is broken with its pressure sensors and the toolhead gantry is bent/misaligned. I don't really know what to do now. Not cool. Seriously.

DoTsTeR77, 2024-02-28:
Please just give us an option to invert it on a per machine basis.

wilk-polarny, 2024-02-28:
> Instead, our icons were designed to clearly show that the z distance will be increasing when tapping the "increase" buttons and that the z distance would be decreasing when tapping the "decrease" buttons.

Wouldn't it be better to have the options:  bed UP and bed DOWN - and then translate according to the printers affected by the action (Either Z+ or Z-) ? That would be a more fool-proof approach, imho

DoTsTeR77, 2024-02-28:
> As mentioned, machines can be used as group masters and therefore the
> logic needs to be applicable to all machines in the group. It would be
> terribly confusing to a user if the logic worked one way when the machine
> is selected as a stand-alone machine and then another way when it is
> selected as a group controller.
> would be decreasing when tapping the "decrease" buttons. The label above
> the buttons also indicates that they are in reference to the Z distance and
> not the bed. On the other hand, the label on the X1 and P1 indicates that
> it is in reference to the bed and the icons clearly indicate that the bed
> will be moving up and down.
> The logic of our implementation is the only way to make it applicable to a
> group of machines.

Many of us don't use groups.  I have a bambu touch on every printer.. could
care less about groups or controlling multiple machines..  I just want to
go to one of my p1s machines and use the big touchpad instead of the awful
one that came with the machine.. and I'd like the up and down movement to
be the same as every other core xy printer I have in my shop. Just telling
you my use case.

mdreizin, 2024-02-28:
@bigtreetech has consistently shown that it values and respects user feedback, which is commendable. However, I noticed that @looxonline closed this particular issue without apparent consideration. I agree with the feedback provided, and as engineers, many of us comprehend the technical aspects involved. Nevertheless, from a user experience standpoint, such actions seem counterintuitive. I kindly suggest reviewing the user's comments on this matter in the Facebook group for a more in-depth insight.

Highlander-bdj, 2024-02-28:
@bigtreetech this issue should be reopened.

looxonline, 2024-02-29:
> Wouldn't it be better to have the options: bed UP and bed DOWN - and then translate according to the printers affected by the action (Either Z+ or Z-) ? That would be a more fool-proof approach, imho

To understand the impact on each machine and how it would lead to a poor user experience, may I ask you to do something like this? Complete the gaps in the headings that I have written below:

Bed Up Button:
Action on P1/X1 -->
Action on A1 -->
Is the nozzle closer or further to the bed on both machines after the action is completed? ________
Is this a consistent behaviour between the two machines? ___________

Bed Down Button:
Action on P1/X1 -->
Action on A1 -->
Is the nozzle closer or further to the bed on both machines after the action is completed? _________
Is this a consistent behaviour between the two machines? ___________

looxonline, 2024-02-29:
> Many of us don't use groups. I have a bambu touch on every printer.. could care less about groups or controlling multiple machines.. I just want to go to one of my p1s machines and use the big [...]

There exists the possibility that for instances where only a single printer is being controlled, we could change the function of the buttons since there is no ambiguity between machines.

However, we have a major concern that we will just see another wave of complaints when we do this because of the reasoning below:

Consider someone controlling multiple machines. At first, he selects only a P1S from the printer selection menu. He moves the bed up and down using the arrows which work the same as the stock X1 screen. He then goes back to the printer selection menu and selects a group. Now, the function of the arrows suddenly changes (because it has to otherwise it won't make any sense for other printers in the group). He hits the key that he thinks means bed up but now it means "increase z height" and the bed drops. Now the button takes on two different behaviours which would be terribly confusing. OK, now let's assume that we kept the button behaviour so that "up" means "up" for all printers in the group. When he hits "Up" the X1 starts to get closer to the nozzle and the gantry on the A1 starts to move up and away from the nozzle. He is only watching the X1 so when he gets the nozzle to the desired height he then glances over to the A1 and notices that the nozzle is actually floating several cm above the bed. What is going on? 

The only way to make these buttons have a consistent meaning is to make them reference an increase or decrease in z height.

DoTsTeR77, 2024-02-29:
Just asking for an option to invert the up and down.. a setting to change
it if the user wants to change it.   Make the default behavior whatever you
like but please give us a way to change it.. thats all.. thank you

DoTsTeR77, 2024-03-02:
This request is not invalid and this issue should not be closed.

looxonline, 2024-03-03:
> This request is not invalid and this issue should not be closed.

It's not that we are against improving this experience for users. We are against degrading it. So far, there has been no approach suggested that would result in an improvement without there being some critical trade off for another user group that would just upset them.

Closing of this issue does not mean that we will not consider trying to improve the user experience in this regard in the future. It just means that the original post was invalid since the controls were not inverted as the OP had suggested. In the context of increasing and decreasing z height, the controls do exactly what they say they do.

If anyone thinks of an implementation which does not impact the another user group in a negative manner and creates an issue to request that implementation then we likely implement it. We are also constantly thinking about this and trying to find a better solution.

DoTsTeR77, 2024-03-03:
Well if you really dont want to make it an option that is easy to switch..
maybe just a way for an advanced user to change the begavior. . for
example.. I don't care if I have to ftp in and change a value is some file.
Then it won't confuse less advanced users as they wouldn't even know about
unless they sought out info on doing it.

By closing the issue you've pretty much made it so no one will see this
thread. The issue should remain open.

Safemode, 2024-03-04:
I would also like to see some sort of opt-in option to reverse the Z-Axis behavior, even if it's in some "secret advanced menu." Such as tapping the 4 corners of the screen in order to access it so it's out of the general view.

looxonline, 2024-03-06:
> I would also like to see some sort of opt-in option to reverse the Z-Axis behavior, even if it's in some "secret advanced menu." Such as tapping the 4 corners of the screen in order to access it so it's out of the general view.

These are definite possibilities. I am going to give this some more thought later today and see what we can do. I can't guarantee anything in the very next update but maybe in the one after that. If we come up with a reasonable solution then I'll create a new issue as a FR and reference this one for the back story.

DoTsTeR77, 2024-04-02:
have you come up with a solution.. for those who are willing to invert the behavior.. thank you..

philchfree, 2024-04-03:
May not be simple but add a switch in options or on the settings page labeled: "printer moving part" where you can select "head" or "bed". When you select "head" the "z" label on the page becomes "head" and up moves the head up, down move the "head" down (A1). If you select "bed", "z" label on the page becomes "bed" and up moves the bed up and down moves the bed down (X1/P1). When you select the switch you can even add a warning "please confirm your printer type". To make it even more clear (but this will have to be updated after any new model by Bambu), instead of "printer moving part" just use "printer type" and switch is "A1" or "X1/P1", rest the same.