bigtreetech/PandaTouch #9: Language support?

UBEXHH, 2024-02-05:
I was hoping for a german (or other) language support

When will this roll out?

Need assistance for free? PM me

looxonline, 2024-02-26:
We may be able to make special language packs that can be "bootloaded" OTA to the unit. We are looking into the feasibility of doing this.

6344-17, 2024-02-26:
Great, let me know if I can help with the GERMAN version :-)

looxonline, 2024-02-28:
V1.0.4 will likely have a JSON that allows key-value pairs to be edited. The values will contain all text used in the UI. Keys must stay the same.

dilixi, 2024-09-14:
> Great, let me know if I can help with the GERMAN version :-)
> […](#)
Would you mind sending me a message 1284513393@qq.com ? 
If you are still willing to help us, we will send you a YML file for translation.

6344-17, 2024-09-14:
Hi JonnySure, looking forward to help...Greetings Frank

dilixi, 2024-09-14:
Hi 6344-17,
This is the yml file, I can't see the emai address.
For example:
We need to change the 'English', and keep the order of the entries.
lang_name_t: 'English'.
Thank you.

[en-GB.zip](https://github.com/user-attachments/files/17003901/en-GB.zip)
@6344-17

dilixi, 2024-11-19:
> I was hoping for a german (or other) language support
> 
> When will this roll out?
> 
> Need assistance for free? PM me

[image]

Hi, 
Panda Touch supports german with V1.0.6.
Please close this issue thanks.

closed 2024-11-20 by BTTJUSTIN

6344-17, 2024-11-22:
Dear colleagues

sorry, we were on longer holidays….
Below is the translation to GERMAN:

￼

Greetings
Frank