bigtreetech/PandaTouch #14: stl/gcode thumbnail while printing?

UBEXHH, 2024-02-06:
The preview of the print is not available while printing.
Why?
Instead it is showing a comic figure of a bear to hide the flaw..

please provide previews of the stl / gcodes that are being printed.

Carias42, 2024-02-09:
Ceci est une bonne idée mais le flux en live serait encore meilleur

suxmcgee, 2024-02-13:
this would be nice.... but at least make the panda animated in the meantime

Bambudidi, 2024-02-15:
Yes, this is what I want too

looxonline, 2024-02-26:
It is not an issue in as much as it is a limitation of the hardware. The thumbs are embedded within the 3mf. This is ok for 3mf files that are just tens of kb large but as soon as things get larger it becomes an issue due to local storage on the panda touch and the need to unzip the thumb from the 3mf. We are looking into ways to display thumbs and if we find a feasible option we will implement it.

philchfree, 2024-03-11:
If you have an agreement with Bambu through the API why not suggest Bambu Lab to modify Bambu Studio to export not only the .3mf in the cache folder of the SD card but also the plate_1.png at the root of the SD card renaming it "name of the 3mf file".png. This way extracting will be done by Bambu Studio and you just have to pick it up to display it in Panda Touch. Benefit for Bambu Lab is that it can use it to display the model in Bambu Studio device tab which is currently not available on LAN Only Mode.

looxonline, 2024-03-12:
> If you have an agreement with Bambu through the API why not suggest Bambu Lab to modify Bambu Studio to export not only the .3mf in the cache folder of the SD card but also the plate_1.png at the root of the SD card renaming it "name of the 3mf file".png. This way extracting will be done by Bambu Studio and you just have to pick it up to display it in Panda Touch. Benefit for Bambu Lab is that it can also use it to display the model in Bambu Studio device tab which is currently not available in LAN Only Mode.

We have actually asked for exactly something like this but at this point communications are strained, likely due to the A1 recall and the stress around that. We can ask but there is no guarantee.

philchfree, 2024-03-12:
> We have actually asked for exactly something like this but at this point communications are strained, likely due to the A1 recall and the stress around that. We can ask but there is no guarantee.

That would be great. Thanks!

3dfixer, 2024-03-16:
@looxonline can we/you not implement it similar to how the BigTreeTech TFT Touchscreen does it?. 
Bambu studio is a clone of prusaslicer

childcarepro, 2024-06-28:
+1 preview image.

In the meantime, could we have the option to display the **file name** in a large font in lieu of the panda?

Derekzielinski, 2024-07-02:
Would love to see this too. That Panda just takes up so much space for no real good reason.

missile150, 2024-07-10:
I agree with all of the above.  Add a jpg of the print and larger text for the file name instead of that panda.

childcarepro, 2024-07-11:
> It is not an issue in as much as it is a limitation of the hardware. The thumbs are embedded within the 3mf. This is ok for 3mf files that are just tens of kb large but as soon as things get larger [...]

I appreciate the limited local storage. Could it be implemented via the flash storage slot?

markcm2, 2024-07-19:
> I appreciate the limited local storage. Could it be implemented via the flash storage slot?

Great question, I was wondering the same; if the removable flash could be leveraged, as I don't use that for any other functions.

looxonline, 2024-07-19:
> Great question, I was wondering the same; if the removable flash could be leveraged, as I don't use that for any other functions.

We've worked up some methods to work around the limited memory. Likely that we will be have to display thumbnails while printing in V1.0.6 and V1.0.5 will show thumbs of all files on the USB drive while browsing.

philchfree, 2024-07-19:
Is there a way to access panda touch through FTP so that we can access the USB key remotely?

looxonline, 2024-07-19:
> Is there a way to access panda touch through FTP so that we can access the USB key remotely?

Not at the moment. If code space allows, it is not impossible. Better to raise it in a separate issue though.

philchfree, 2024-07-19:
Done, thanks

dilixi, 2024-11-19:
@

> The preview of the print is not available while printing. Why? Instead it is showing a comic figure of a bear to hide the flaw..
> 
> please provide previews of the stl / gcodes that are being printed.
> 
> i want this:
> 
> ![IMG_1705](https://private-user-images.githubusercontent.com/150461275/302550216-34419f91-db75-4d4a-849b-4deb6ba90281.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzIwMDY2MzAsIm5iZiI6MTczMjAwNjMzMCwicGF0aCI6Ii8xNTA0NjEyNzUvMzAyNTUwMjE2LTM0NDE5ZjkxLWRiNzUtNGQ0YS04NDliLTRkZWI2YmE5MDI4MS5qcGc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQxMTE5JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MTExOVQwODUyMTBaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT04ZmI1ZGY2NDE3ZDk3OWZhYmY1Y2EwM2ZmOWNmZmQ2MDlhMWU0NTAzN2U2YTlkZmU5MGQyMzVlNmUxMjNlOTg1JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.c1HYNcSmOGVsj3fKutl06iGdUaAJ-uKBoAGi1cUuZns)

Hi,
there are some case which allow the Panda Touch supports the thumbnail on home page,please close this issue ,thanks.

[image]

BTTJUSTIN, 2024-11-20:
This bug was fixed in FW version V1.0.6

closed 2024-11-20 by BTTJUSTIN

FireBall1725, 2024-11-22:
I upgraded to v1.0.6 and i still see the panda image instead of the preview image, i've tried printing 10 different things, and none of them showed the preview. Is there something that has to be done to enable this feature?

dilixi, 2024-11-22:
> I upgraded to v1.0.6 and i still see the panda image instead of the preview image, i've tried printing 10 different things, and none of them showed the preview. Is there something that has to be done to enable this feature?

Please make sure that:
1 the print task is started by bambu studio
2 the thumbnail can be loaded on sd card panel.

If there is nothing change you can use the V1.0.6 which is placed in my rep.

https://github.com/dilixi/PandaTouch/tree/master/Firmware/1.0.6

dilixi, 2024-11-22:
> Please make sure that: 1 the print task is started by bambu studio 2 the thumbnail can be loaded on sd card panel.
Hi,
@FireBall1725 

 If there is nothing change you can use the V1.0.6 which is placed in my rep.
> 
> https://github.com/dilixi/PandaTouch/tree/master/Firmware/1.0.6

FireBall1725, 2024-11-22:
@dilixi, i downloaded the v1.0.6 firmware from your link above and flashed my Panda Touch, however i'm still seeing the panda image and not the image of what i'm printing

dilixi, 2024-11-22:
> @dilixi, i downloaded the v1.0.6 firmware from your link above and flashed my Panda Touch, however i'm still seeing the panda image and not the image of what i'm printing

Did you start a new print task from bambu studio?

Please make sure that:
1 the print task is started by bambu studio
2 the thumbnail can be loaded on sd card panel.

FireBall1725, 2024-11-22:
> 1 the print task is started by bambu studio
> 2 the thumbnail can be loaded on sd card panel.

Yep, I started it from in the Bambu Studio, and I also can see the thumbnail from the panel when looking at whats on the sd card.

dilixi, 2024-11-22:
> > 1 the print task is started by bambu studio
> > 2 the thumbnail can be loaded on sd card panel.
> 
> Yep, I started it from in the Bambu Studio, and I also can see the thumbnail from the panel when looking at whats on the sd card.

What is the firmware version of the printer? And what is the type of it?x1?
Would you mind send me the .3mf file via email(1284513393@qq.com).

FireBall1725, 2024-11-22:
I have a Bambu X1C, the firmware is shows is 1.08.00.00 because i'm running X1 Plus Firmware, which is on version 2.0.

I will send you the 3mf

dilixi, 2024-11-22:
> I have a Bambu X1C, the firmware is shows is 1.08.00.00 because i'm running X1 Plus Firmware, which is on version 2.0.
> 
> I will send you the 3mf

[image]

X1 is different,but I will test it with that .3mf file.

FireBall1725, 2024-11-22:
ahh, thank you!! sorry I didn't catch that part.

koffienl, 2024-12-15:
> Did you start a new print task from bambu studio?
>
> Please make sure that: 1 the print task is started by bambu studio 2 the thumbnail can be loaded on sd card panel.

I am clearly doing something wrong, but do not know what.

- Brand new panda touch
- upgraded to v1.0.6.1
- P1S with firmware 01.07.00.00
- Factory reset after update

When browsing the 3MF files through panda touch I simply do not see any thumbs, just empty squares. As result I also don't see a thumb on the homepage.
I'm using Bambu Studio 1.10.1.50, all files are sliced/send through the studio.

[image]
[image]

markcm2, 2024-12-15:
> > > @dilixi, i downloaded the v1.0.6 firmware from your link above and flashed my Panda Touch, however i'm still seeing the panda image and not the image of what i'm printing
> > 
> > 
> > Did you start a new print task from bambu studio?
> > Please make sure that: 1 the print task is started by bambu studio 2 the thumbnail can be loaded on sd card panel.
> 
> I am clearly doing something wrong, but do not know what.
> 
>     * Brand new panda touch
> 
>     * upgraded to v1.0.6.1
> 
>     * P1S with firmware 01.07.00.00
> 
>     * Factory reset after update
> 
> 
> When browsing the 3MF files through panda touch I simply do not see any thumbs, just empty squares. As result I also don't see a thumb on the homepage. I'm using Bambu Studio 1.10.1.50, all files are sliced/send through the studio.
> 
> ![20241215_172202](https://private-user-images.githubusercontent.com/6832264/395877273-b3491e45-7d8a-4ea0-8f51-e89937988490.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzQyODQyNDQsIm5iZiI6MTczNDI4Mzk0NCwicGF0aCI6Ii82ODMyMjY0LzM5NTg3NzI3My1iMzQ5MWU0NS03ZDhhLTRlYTAtOGY1MS1lODk5Mzc5ODg0OTAuanBnP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MTIxNSUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDEyMTVUMTczMjI0WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9NTMyMzNjMzY5NjUxZDU3OTM1YTE0OTMyMjQzYWM5NTRhN2Y4ODcxZTE4ZDUxYzRlZDk4MTRiODcyMWZkYTQwMiZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QifQ.qPEFy1dpUp6zJr5eD4X2DpDJnH9_KTBDut9-DhEaphI) ![20241215_172212](https://private-user-images.githubusercontent.com/6832264/395877276-2399223b-a69e-44fc-a4b6-db4eaf41fe94.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzQyODQyNDQsIm5iZiI6MTczNDI4Mzk0NCwicGF0aCI6Ii82ODMyMjY0LzM5NTg3NzI3Ni0yMzk5MjIzYi1hNjllLTQ0ZmMtYTRiNi1kYjRlYWY0MWZlOTQuanBnP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MTIxNSUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDEyMTVUMTczMjI0WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9ZWM4YjU0MWU5MWMwYTVmNjEzMGFmNzdkODlkNWE0ZDc2MDg5OWY4ZDhkMTAzOGRhZjVkZTM2MjM2YWZkZjg2MiZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QifQ.LiWRLAXOmhjSdiETMoy8RUhobi6cKuC-rLd70ua0UTg)

This stumped me as well, and I could not find any information on this and almost gave up. Eventually I found that by clicking the gear icon (forth icon down on the left, and then clicked Enhanced Icon On, and selected the ON option, then icons began showing. 

Assuming that the default setting of NO was the reason icons did not show, it blows my mind that this is not mentioned anywhere and part of the instruction.

Maybe that's not it and I just dont understand how these images work but its worth giving a try.

BlueFalconOne, 2024-12-19:
Same setup as @koffienl

markcm2, 2024-12-19:
Another odd characteristic I noticed- my thumbnails stopped working recently, I opened settings and checked that the enhanced thumbnails were still on, then opened the file browser which has a refresh button and clicked that, then on the home page, I double tapped on the image of the panda. I don’t know which of these steps were responsible but as soon as I double tapped on the panda, the image refreshed with the image of the part I was printing.

BlueFalconOne, 2024-12-19:
@markcm2 I just tried that.  Still nothing on my end.  Just a Panda chowing down on some bambu.

BlueFalconOne, 2024-12-19:
OK scratch all my complaints...  on my next print, the icons started showing.  No clear reason why, so I would try what @markcm2 suggested and hopefully it starts working again for everyone!

ChristianBracket, 2025-04-19:
Still not working for me. P1S, panda touch with firmware 1.0.7, nothing works as far as  the thumbnails go. Any help would be lovely!