bigtreetech/PandaTouch #16: Dynamically map filaments at the start of the print.

alexandrkut, 2024-02-06:
Very necessary options:
Dynamically map filaments at the start of the print.

Gizmo6, 2024-02-18:
This is essential. Would have thought it would be present from day one! Bump

looxonline, 2024-02-28:
Non-trivial to implement. The 3mf requires analysis to determine the current mapping. This would require the 3mf to be downloaded to the panda touch, uncompressed, analysed and then for the current filament mapping to be displayed. It would be essential for the panda touch to have a USB flash inserted for this process to be successful since there is not much local memory. Furthermore, large 3mf files may take a while to download to the panda touch. While this is not ruled out, it will not appear in the next two firmware releases where more pressing enhancements will be addressed.

looxonline, 2024-03-12:
> Agree this should be a high priority. Would 1 filament files still need analyzed? If not, having this feature only functional for 1 filament jobs (in the short-term) would be valuable.

We are actually looking into being able to do this for just single filament prints. Won't be in the next release but it is under investigation for the one thereafter.

RyanEwen, 2024-04-19:
If it helps, this project supports this in their batch print mode when uploading gcode files: https://github.com/TFyre/bambu-farm/tree/main