bigtreetech/PandaTouch #21: Request - Support for folders and .gcode files

AndiS977, 2024-02-08:
Please add support for (sub)folders and .gcode files on the printers SDcard.

CyberHugger, 2024-02-15:
Please!

looxonline, 2024-02-26:
gcode files are not supported by the BBL machines. It must be in 3mf format. Folders is already under consideration and mentioned in another issue report. Closing as a duplicate.

closed 2024-02-26 by looxonline

AndiS977, 2024-02-27:
gcode files **are** supported. I can export as gcode files and then transfer them via ftp to the machines SD card (P1P). Then I can select them on the stock display to print. It has the advantage to start the print faster, because the embedded gcode has not to be extracted out of the 3mf.

looxonline, 2024-02-27:
> gcode files **are** supported. I can export as gcode files and then transfer them via ftp to the machines SD card (P1P). Then I can select them on the stock display to print. It has the advantage to start the print faster, because the embedded gcode has not to be extracted out of the 3mf.

They are not supported when initiating a print over MQTT.

AndiS977, 2024-02-27:
Ah, ok. Thank you for your answer.