bigtreetech/PandaTouch #22: Add a battery level indicator to top

Qjeezy, 2024-02-09:
Currently I cannot see how much battery there is left on the unit when I take it to go. It’s a guessing game of how many minutes I have left. 

A little graphic or percentage at the top next to the wifi indicator would be nice. Not sure if this would also require a hardware upgrade or not though.

closed 2024-02-26 by looxonline

suxmcgee, 2024-02-26:
Odd - the ESP32-S3 in the panda touch allows for monitoring via the ADC. Many folks have written code to use this for battery monitoring.

looxonline, 2024-02-26:
> Odd - the ESP32-S3 in the panda touch allows for monitoring via the ADC. Many folks have written code to use this for battery monitoring.

It does but every single pin is used up. We had a design decision to allow control of screen brightness or battery level and we selected the former.

oculardynamic, 2024-02-26:
My .02 cents, I thinkScreen can be tinted externally pretty easy. I designed a screen bezel for the device and thinking about it, would be pretty easy to make it a 2 piece design you can snap a piece of window tint or something in to for getting a dimmer panel. Yeah it a little more work, but I'm not sure how you'd add a battery icon externally lol. Just saying you'd then have access to having both

Carias42, 2024-02-26:
If I can choose a prefer 1000X an battery indicator

looxonline, 2024-02-26:
> > > > > Odd - the ESP32-S3 in the panda touch allows for monitoring via the ADC. Many folks have written code to use this for battery monitoring.
> > > > 
> > > > 
> > > > It does but every single pin is used up. We had a design decision to allow control of screen brightness or battery level and we selected the former.
> > > 
> > > 
> > > Ability to see the battery level, especially while operating on battery, seems a lot more important than screen brightness. I’d be more interested in knowing how long I have left on my battery than changing the screen brightness. One is actually useful. Just set reasonable screen brightness levels.
> > 
> > 
> > You'd be surprised how important many consider screen brightness to be. This decision was made after taking into consideration a lot of feedback from beta testers.
> 
> Maybe take feedback from the wider audience now versus the few folks that tested it out before release and likely received their device for free and were sugar coating it.

The issues section on this repo exists for the sole purpose of getting feedback. We are most certainly taking feedback and suggestions. Ideally, both could be implemented in a future hardware revision but it may mean some tricky multiplexing of some pins or giving up another feature. It is something that we are considering for the future.

bcarriveau, 2024-02-26:
> The issues section on this repo exists for the sole purpose of getting feedback. We are most certainly taking feedback and suggestions. Ideally, both could be implemented in a future hardware [...]

Maybe it could be added to a dock, with more batteries.. hint hint, wink, wink.

looxonline, 2024-02-26:
> Maybe it could be added to a dock, with more batteries.. hint hint, wink, wink.

I like it. Could report the battery voltage of the dock via I2C. Will definitely be considered.