bigtreetech/PandaTouch #23: Allow master/slave printer groups

Qjeezy, 2024-02-09:
I’d like to have more than one master printer with each their own slave printers. This way we can control more than one group of printers independently from the other groups of printers. Hope that makes sense.

papsg, 2024-02-20:
If implemented please use modern language such as "controller/worker" or something else that isn't "master/slave". The tech industry is slowly moving away from that language.

looxonline, 2024-02-26:
Coming in V1.0.3

dilixi, 2024-11-19:
> I’d like to have more than one master printer with each their own slave printers. This way we can control more than one group of printers independently from the other groups of printers. Hope that makes sense.

@Qjeezy 

Hi,
Please close this issue, because it was implemented on V1.0.3. Thanks.

closed 2024-11-19 by Qjeezy