bigtreetech/PandaTouch #24: Request - Add "Sort by" or allow tapping column header name to sort USB files.

Vantage-DS, 2024-02-09:
Currently, files on the USB seem to be in Date order (or "Data" order as the display shows) 🙃

Allow tapping the column name header to sort by that header, descending or ascending on sequential taps.

looxonline, 2024-02-26:
Will be considered for V1.0.3 release.

Vantage-DS, 2024-04-03:
Will this be implemented in 1.0.4 now? I really think it's necessary at this point.
I've got 77 files that are in no discernable order and as I refine new printer settings and re-add the gcode to the Panda USB stick, the order becomes more and more chaotic.

philchfree, 2024-04-03:
I would prefer a folder implementation

superceleron, 2024-04-03:
> I would prefer a folder implementation

Same here

Vantage-DS, 2024-04-03:
So would I, but that's another feature request... Not this one.

superceleron, 2024-04-03:
btw, is already a 1.0.3 fw out that i do not know about? since i cant find it.

Vantage-DS, 2024-04-03:
> btw, is already a 1.0.3 fw out that i do not know about? since i cant find it.

If you go on the Facebook 'Bigtreetech Panda Touch User Group', there's a post in there you can put your GitHub ID down and they'll add you to the beta test github.

superceleron, 2024-04-03:
> > btw, is already a 1.0.3 fw out that i do not know about? since i cant find it.
> 
> If you go on the Facebook 'Bigtreetech Panda Touch User Group', there's a post in there you can put your GitHub ID down and they'll add you to the beta test github.

Thks :)

BTTJUSTIN, 2024-09-13:
This feature was implemented in FW version V1.0.5

closed 2024-09-13 by BTTJUSTIN