bigtreetech/PandaTouch #25: Manual Pause - Load / Unload greyed out

UBEXHH, 2024-02-10:
When you manually pause a print (or the gcode has a M400 U1 pause) Panda Touch does not let you LOAD or UNLOAD the filament. You have to use the original printer interface to get it working.

Strangely the messages do appear (pull out or Done/Retry after loading)

Please fix

The number of issues is rising but I do not see any activity from bigtreetec. No categorization, reaction, first thoughts, prioritising, dispatching, help, support

DoTsTeR77, 2024-02-10:
Ya.. that occurred to me too.. not a peak from them... kinda figured at
least 1 firmware update by now fixing a few things.. then again it's
Chinese new year till the 16th.. so there's that.. but doesn't some ofvteam
operate out of south Africa.  Idk.

UBEXHH, 2024-02-21:
Before or after the 16th... nothing happens. No reaction, no classification, no priorisation, nothing. Sad.

looxonline, 2024-02-26:
@DoTsTeR77 I was also on vacation during the shutdown.

@UBEXHH We immediately got to work on the PT after returning but there were other priorities that it took precedence over the new firmware release. This issue has been logged for consideration for the V1.0.3 release.

dilixi, 2024-11-19:
> When you manually pause a print (or the gcode has a M400 U1 pause) Panda Touch does not let you LOAD or UNLOAD the filament. You have to use the original printer interface to get it working. [...]

@UBEXHH 
Hi,
This was implemented on V1.0.3,Please close this issue,thanks.

BTTJUSTIN, 2024-11-20:
This bug was fixed in FW version V1.0.3

closed 2024-11-20 by BTTJUSTIN