bigtreetech/PandaTouch #27: Use inclusive language

andypiper, 2024-02-12:
Per https://inclusivenaming.org/word-lists/tier-1/_master-slave/

The terminology of master/slave is outdated, please could this be updated to something more modern such as primary/secondary, or controller/secondary, or similar.

Thanks! 💟

papsg, 2024-02-23:
+1
This terminology is super cringe and the entire tech industry is moving away from it.

nescaffier, 2024-02-26:
+1 Primary/Secondary is the preferred term

chuckkay, 2024-02-26:
frail individuals.... cry in a corner and shiver, no way to go through life....   if you can't handle words used with no bad intentions, good luck enjoying life... lets just burn all the classic comedy movies from the 80's while we're at it....   Give me a break.....

Kooties, 2024-02-27:
> frail individuals.... cry in a corner and shiver, no way to go through life.... if you can't handle words used with no bad intentions, good luck enjoying life... Give me a break..... Group think snowflakes....... where does it end??? lets just burn all the classic comedy movies from the 80's while we're at it....

oh, someone's big mad about people using more inclusive language

looxonline, 2024-02-27:
This concept is quite foreign to us. The principle of master/slave is common in electronics and has been for many decades but we have not heard of this before. It seems that some are drawing a parallel to literal slavery and this is where the issue is? This seems strange since it seems that it was mostly in the USA that slavery was enforced in a very harsh manner and this is where the concern stems from but electronics is used globally. In other lands the literal master/slave arrangement could be a very positive one with some slaves choosing to remain with their master even after they could afford to purchase their freedom. Nevertheless, if there are people who have been affected by the literal arrangement in the USA who are hurt by these terms then we can rename them.

papsg, 2024-02-27:
> This concept is quite foreign to us. The principle of master/slave is common in electronics and has been for many decades but we have not heard of this before. It seems that some are drawing a [...]

Thank you for adding this as a milestone.

DoTsTeR77, 2024-02-27:
I vote you leave them as they are.. the whole thing is ridiculous.  Soon as
you change them some snowflake will be probably be offended by the new
terms.

andypiper, 2024-02-27:
As the user that raised this issue, I'll point back to the website I linked in my first post, inclusivenaming.org - I am not affiliated with that site, but agree with the goal (beyond this specific term in this specific project): "Helping projects and companies make consistent, responsible choices to remove harmful language". I'm not hurt by anything here, I'm making a suggestion to help the product be more modern and approachable.

This is an ongoing effort in the technology industry, that has been going on for many years, and has seen (for example) GitHub and git switch from "master" to "main" terminology.

I appreciate BigTreeTech taking a look into this, with the goal of modernising the user experience of the product.

chuckkay, 2024-02-28:
Woke Losers...... Group Think morons..... Grow the fuck up...... Dare I say it.... Yes..screw it, I will......... your a bunch or Libtards.....

looxonline, 2024-02-28:
It seems like people are divided about this and it is OK to have a different opinion but I had to delete a comment that was quite aggressive which is not right. Please keep the discussion friendly.

Vantage-DS, 2024-02-29:
Personally, I grew up with the term Master/Slave and immediately think of an IDE hard drive with jumper configs from days of old.

Whilst calling people snowflakes probably isn't the best way to enter a debate, IMO keeping master/slave is fine. There is nothing wrong with calling a piece on electronic equipment this.

USA might hold the torch for 'popular' slavery, but it's happened everywhere on earth throughout time, and still continues in many countries today. Renaming electronic terminology will do nothing to impact this, I really can't see an issue 

That said, times change, words change. Is up to BTT what they want to call it now.

DoTsTeR77, 2024-02-29:
Yes ide hard drives for sure.. that was the 1st thing I thought of as
well..  as for the snowflake comment you should have seen that post that
got deleted.. that guy didn't hold back.

heelsmj, 2024-03-24:
> Per https://inclusivenaming.org/word-lists/tier-1/_master-slave/
> 
> The terminology of master/slave is outdated, please could this be updated to something more modern such as primary/secondary, or controller/secondary, or similar.
> 
> Thanks! 💟

I cannot believe people actually take time to write something like this. Get a life. Stop going through life looking for every perceived slight

Vantage-DS, 2024-04-09:
This can probably be closed. The word 'Leader' is now being using in the beta.