bigtreetech/PandaTouch #29: Use of i2c connector

andypiper, 2024-02-12:
It looks like the i2c connector is a Grove socket - when will this be enabled?

jlo1975, 2024-02-14:
I've received the panda touch screen and all is fine.
I'm trying to use a DHT20 humidity and temperature sensor with digital I2C output as there is a port on the back and we can see a graphical placement in the home screen.
But it seems the panda touch devise does not see it. I've tested the wiring and all is ok (tested 3.3v ).

What am I missing ? Is the external sensor with digital I2C is supported or is it planned in the futur?

looxonline, 2024-02-26:
Coming in a future release. Closing for now.

looxonline, 2024-03-28:
Reopening since we are starting to plan for this in V1.0.4

looxonline, 2024-07-18:
V1.0.4 shifted focus to cloud mode because of the changes that Bambu implemented. V1.0.5 will be focused on the Panda PWR and thumbnail support. V1.0.6 will add I2C sensor support.

BTTJUSTIN, 2024-11-20:
This bug was fixed in FW version V1.0.6

closed 2024-11-20 by BTTJUSTIN