bigtreetech/PandaTouch #31: [Initial initialization] Cannot connect to network

TheOrdner, 2024-02-13:
After powering up the Panda Touch the first time, trying to connect to network does not work. Always sates

 "Failed to connect to current WiFi, Please check signal strength & password entered and then reconnect."

Standing right next to the router. 2.4GHz network. No special characters in SSID nor password. Power cycling does not help.

bcarriveau, 2024-02-17:
It's something unique to your SSID or password, most likely. You appear to be the only one with an issue out of thousands of units.

TheOrdner, 2024-02-17:
I certainly can assure you that it has nothing to do with my SSID nor password. Only letters and numbers are being used and no whitepsaces.
There are others on Reddit too. It seems to have something to do with WPA3 but thanks for looking into that.

bcarriveau, 2024-02-17:
Can you set up a less secure guest network and try to connect? Maybe just
WEP or WPA2? Then provide specific details for them to troubleshoot with.
You didn't provide anything helpful for the developers.

DoTsTeR77, 2024-02-17:
Mine wouldnt connect yesterday after turning the printer and bambu touch on
after both being powered off.. it just took a few power cycles of the bambu
touch and and it finally connected. It's the 1st time it happened to me. I
just assumed the printer wasn't quite setup on the network and ready yet..
it has been fine since.m

DoTsTeR77, 2024-02-17:
> Haven't found time to look further into it, because I bought a cheap
> access point so its working for now. Will update once I find some time

Jinxed myself.  Turn everything on an off again and now the bambu touch
won't connect to my p1s.. it connects to my a1 just fine still though..
 so I'd say this person isn't the only one seeing this.

bcarriveau, 2024-02-17:
> Jinxed myself. Turn everything on an off again and now the bambu touch won't connect to my p1s.. it connects to my a1 just fine still though.. so I'd say this person isn't the only one seeing this.
> […](#)

The OP in thisnissues 

> Jinxed myself. Turn everything on an off again and now the bambu touch won't connect to my p1s.. it connects to my a1 just fine still though.. so I'd say this person isn't the only one seeing this.
> […](#)

The OP couldn't connect to his Wi-Fi network but left no details about it.

Your self-inflicted wound is allowing your printer to run on DHCP leases, causing you issues, not a PandTouch issue. Either reserve the IP in your router or manually assign it one outside the DHCP pool.

DoTsTeR77, 2024-02-17:
Already assigned an ip in a router since it happened but thanks for the
lecture there sizzle chest..

TheOrdner, 2024-02-17:
Im no expert when it comes to networks so I dont really know what details are relevant and Iam happyto share everything that is important to this matter.

looxonline, 2024-02-26:
@TheOrdner  Were you able to get it to connect at all? There is something unique about your network that we would love to figure out.

netmagi, 2024-03-07:
I also cannot get mine to connect to wifi.  Same issue as OP.  Have not even been able to complete setup.

My access points are Netgear.  I have a WAX214PA, and a WAX218.

netmagi, 2024-03-07:
> I also cannot get mine to connect to wifi. Same issue as OP. Have not even been able to complete setup.
> 
> My access points are Netgear. I have a WAX214PA, and a WAX218.

ok, I did actually just get it to connect finally after power-cycling the panda touch.  Maybe it's a weird first-boot issue?  I tried over and over again before power-cycling it and no-go, but it actually connected on it's own when I turned it off-on.

ShRTee, 2024-04-12:
I just bought a Panda Touch and i have two differnt 2.4 Ghz networks and i use to work setting networks up.
There is a big issue with the wifi on the touch.

I have a xtouch screen set up and no issues with that.

I bought the Touch for having the extra functions.

Im now setting up a specific 3D network on 2.4 for the touch.
I will post updates ..

insp1re1, 2024-05-09:
same new device, cant connect to wifi, using unifi ac lr

MacMcDell, 2025-07-01:
every damn time i turn on the printer and panda. ..failed to connect to network. cycles for about 200 seconds.. restart.. finds it .. connects.. failed to connect to network. just a massive PITA. asus RT-AX58U. Mapped both mac address for bambu printer and the panda touch.. no connection. it shows up on my network after restart.. then promptly drops. 

[image]