bigtreetech/PandaTouch #33: [WIFI] painful process of setting wifi on Panda Touch and bugs related to it

arekm, 2024-02-17:
I have a wifi network with multiple (6) access points that are using the same SSID name "X". That's neat setup where mobile wifi devices can automatically roam between APs.

Problems I noticed:

1. PandaTouch firmware shows each Access Point as separate network (despite all APs using the same SSID)

2. above can be ok if esp32 doesn't support roaming and needs to associate to particular single AP

3. but when listing networks please show BSSID of each AP in such case so I'll know which AP I'm telling Panda Touch to connect

4. same for clicking on already connected network - show AP BSSID alongside of currently shown IP and MAC address

5.  also please change graphic for signal strength to have more bars (like 6) or just numerical % value. That way we can differentiate APs. Currently it shows all my APs with two bars where one AP is just above a printer and other AP is in 2 rooms away and the signal *is* different but I can't deduce that from what Panda shows me.

6. make it possible to forget network (currently there is no way to do that)

7. currently Panda shows each AP with the same SSID "X" as separate network but clicking on other entry in list than currently connected (so another AP but for the same "X" SSID) causes nothing. It should ask for password and connect to newly clicked AP (which has the same SSID as current AP, as mentioned earlier).

looxonline, 2024-02-26:
Improvements to this coming in the next firmware

looxonline, 2024-02-28:
The PT will always attempt to connect to the SSID with the highest RSSI is more than one SSID with the same name in the new firmware.

closed 2025-09-29 by xperiments