bigtreetech/PandaTouch #34: Implement Folders.

Gizmo6, 2024-02-18:
If the intention of the panda touch is to help with the running of print farms. Then thumbnails and FOLDERS are essential.

looxonline, 2024-02-26:
Coming in next firmware.

Vantage-DS, 2024-06-15:
Bumpety bump.

Folders would be welcomed immensely.

I'm currently having to scroll through 100+ files on the USB each time I set prints going 😄

TechnoSwiss, 2024-07-25:
Would this also allow the ability to see gcode files in folders on the SD card? Doesn't seem to be an API limitation as this Octoprint plugin displays and can print from files in the cache directory on the SD card. https://github.com/jneilliii/OctoPrint-BambuPrinter

bigtreetech, 2024-07-25:
Yes, when implemented.

dilixi, 2024-11-19:
> > > If the intention of the panda touch is to help with the running of print farms. Then thumbnails and FOLDERS are essential.
> > 
> > 
> > ![image](https://private-user-images.githubusercontent.com/56668660/387574588-172d5a1e-d7f6-4784-880c-da50cf12d54b.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzIwMDkwOTUsIm5iZiI6MTczMjAwODc5NSwicGF0aCI6Ii81NjY2ODY2MC8zODc1NzQ1ODgtMTcyZDVhMWUtZDdmNi00Nzg0LTg4MGMtZGE1MGNmMTJkNTRiLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDExMTklMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQxMTE5VDA5MzMxNVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTA2MzdlYzc4MWUxMTdhMDQxYWFiNmZkMWE3YWUxY2Q5ZTViNDM1NzdjMDk1ZjI3NzIxNjRlODNlZjBlYmFhYzUmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.1NLFNxzqxZkLcLAoh9xRth5gn-V0kYTlVEtIFSY_WrM)
> > @Gizmo6
> > Hi, This was implemented on V1.0.6,Please close this issue,thanks.
> 
> I just checked and the latest firmware is V1.0.5.1. Or am I missing something? I'm really waiting for folders.

Hi,
You can get it from PR.

glodny29, 2024-11-19:
> > > > If the intention of the panda touch is to help with the running of print farms. Then thumbnails and FOLDERS are essential.
> > > 
> > > 
> > > ![image](https://private-user-images.githubusercontent.com/56668660/387574588-172d5a1e-d7f6-4784-880c-da50cf12d54b.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzIwMDkwOTUsIm5iZiI6MTczMjAwODc5NSwicGF0aCI6Ii81NjY2ODY2MC8zODc1NzQ1ODgtMTcyZDVhMWUtZDdmNi00Nzg0LTg4MGMtZGE1MGNmMTJkNTRiLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDExMTklMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQxMTE5VDA5MzMxNVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTA2MzdlYzc4MWUxMTdhMDQxYWFiNmZkMWE3YWUxY2Q5ZTViNDM1NzdjMDk1ZjI3NzIxNjRlODNlZjBlYmFhYzUmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.1NLFNxzqxZkLcLAoh9xRth5gn-V0kYTlVEtIFSY_WrM)
> > > @Gizmo6
> > > Hi, This was implemented on V1.0.6,Please close this issue,thanks.
> > 
> > 
> > I just checked and the latest firmware is V1.0.5.1. Or am I missing something? I'm really waiting for folders.
> 
> Hi, You can get it from this rep.
> 
> https://github.com/dilixi/PandaTouch/tree/master/Firmware/1.0.6

Thank you, it's working perfectly!

closed 2025-03-22 by dilixi