bigtreetech/PandaTouch #38: Print queue

DanielEikelenboom, 2024-02-21:
To enable continuous printing, some sort of printing queue is a must-have feature for my printfarm. 

I now paste gcode files behind eachother, but gcode is not supported by the PandaTouch. I thus cannot use my PandaTouch as I intented. Now I still have to update 16 SD cards every time I change something to the print file. (My printers auto eject)

sanctas, 2024-02-23:
Could you give more info on how your auto ejection works?

DanielEikelenboom, 2024-02-23:
I added some custom gcode to the end gcode. Basically it just waits for the bed to be cool (with all fans on), then gently sweeps the item of the bed. Tuned for one specific thing I have to print lots of.

However, from bambu slicer, I can only print a single item. So I export the gcode of a single item and paste that a couple of times (enough to empty my AMS). Then I can leave it over the weekend without have to start a new print every 4 hours.

Let me know if you need more information!

looxonline, 2024-02-28:
I like the idea. Will be kept under consideration.

redheadgiraffe, 2024-02-29:
When I select a file on the SD card in Panda Touch and start printing, the filament selection in AMS does not appear. It appears in Bamboo Handy App and Studio, and I hope Panda Touch gets the same update.

maple-underscore, 2025-01-22:
Any update on the print queue feature? I would also like it since I am considering starting a small business.

DanielEikelenboom, 2025-02-03:
There are alternatives with print queues, although not all (fully) free:

https://printandgo.tech/
https://www.printago.io/
https://simplyprint.io/

An update would be much appreciated