bigtreetech/PandaTouch #46: Temperature sensor integration.

maxxe88, 2024-02-28:
I have placed a temperature sensor in my 3D printing space. I have integrated the sensor into my ioBroker. It would be great if you could send the data to the Panda Touch, thus having a temperature display on the screen without needing an X1C. Integration of the AMS would be nice as well. I have a humidity and temperature sensor there. It would be great to integrate this data into the Panda Touch.

looxonline, 2024-02-28:
Integration with other brokers is a challenge as their topics and data formats vary. Code space and TCP sockets on the Panda Touch are limited too. Nevertheless, this is not ruled out but just rather something that will not fall into the next 2-3 firmware releases as there are some very high priority, high impact items already in the queue.

maxxe88, 2024-02-28:
I understand that well. It would indeed be a useful feature. I hope that such integrations will be considered in future developments and can provide you with the desired capabilities.

Thank you

xperiments, 2025-09-29:
Closing this issue as stale due to inactivity.

closed 2025-09-29 by xperiments