bigtreetech/PandaTouch #48: Battery charging indicator

philchfree, 2024-03-12:
I understand that battery level might be difficult to implement but can we just have an icon which shows that the battery is actually charging or that external 5V is received? When I set up panda touch to battery and attach it to the dock I am never sure it works because it is just depleting its battery or it is powered by the dock.

thiagolocatelli, 2024-04-12:
This feature needs to be added.

ideaalab, 2024-07-26:
May I know why its difficult to implement a battery level indicator?
Battery lever is a must!
This board is equipped with an ESP32 microcontroller, that is perfectly capable of showing battery level.

bigtreetech, 2024-07-26:
Unfortunately the TFT eats all of the IO needed for this. An RGB TFT is extremely IO hungry. We will look at the possibility of adding it using an alternate method in the future but at the moment there is nothing that can be done in software as there is no mechanism to measure it.

closed 2024-07-26 by bigtreetech

ideaalab, 2024-07-26:
Oh, what a pity. The most basic Microchip uC allows to read internal ADC for supply voltage readings without occupying any I/O pin. I thought an ESP32 would have something similar.

bigtreetech, 2024-07-26:
> Oh, what a pity. The most basic Microchip uC allows to read internal ADC for supply voltage readings without occupying any I/O pin. I thought an ESP32 would have something similar.

No. All external voltage readings would always require an IO pin. There is no physical way around this for any uC. The only exception would be if someone wanted to measure the voltage that is being supplied directly to the uC but I am yet to see a uC that offers an accessible measurement of that through an internal register other than a rough estimate for brownout detection.

ideaalab, 2024-07-26:
There is a reliable way to do it. We use this for PIC10/12/16 families every time. This microcontrollers have a fixed voltage reference (FVR) selectable 1.024/2.048/4.096v.
You can read this internal reference with an internal ADC. No pin needed.
Let say you configure the Vref to be 1.024v. And you use a Li Ion battery to power your device (nominal 4.2v).
When you read the FVR you will get different ADC values depending on the battery voltage.
But this is possible because this microcontrollers work from 1.8 to 5.5v. Meaning that they can be powered directly by a Li ion battery (no need for a voltage regulator). And because they have an internal ADC channel that allows to read the FVR.
Im checking and its very strange that ESP32 does not have this functionality.

ideaalab, 2024-07-26:
We even have a library for this :)
https://github.com/ideaalab/battery_level_library

TMillesich, 2024-07-26:
> There is a reliable way to do it. We use this for PIC10/12/16 families every time. This microcontrollers have a fixed voltage reference (FVR) selectable 1.024/2.048/4.096v. You can read this [...]

This would need a new hardware revision, which is not the topic of this ticket. ;)

ideaalab, 2024-07-26:
> No. All external voltage readings would always require an IO pin. There is no physical way around this for any uC.

Sure, I was just saying it possible with other uC ;)

TMillesich, 2024-07-26:
> Unfortunately the TFT eats all of the IO needed for this. An RGB TFT is extremely IO hungry. We will look at the possibility of adding it using an alternate method in the future but at the moment there is nothing that can be done in software as there is no mechanism to measure it.

So you're using a 16bit parallel bus for the TFT with an ESP32S3R8. Okay.

Let's hope there'll be a new Panda Touch product version in the future.

garethky, 2026-01-06:
Does it have a charge controller?
How do we know when its charging?
How do we know when its dead?
Will it over discharge the battery if left unattended?
Is it a fire hazard?