bigtreetech/PandaTouch #53: Better utilize screen real estate.

heelsmj, 2024-03-24:
While printing, the two largest areas of the screen are taken up by the panda, and a picture of the currently selected printer. I understand you mimicked the interface from the X1 carbon however, the fact that you’re not displaying a 3-D image of the model currently being printed just makes the whole interface during the print jumbled.  The relevant information as far as the print progress and what’s being printed is relegated to a tiny font on the bottom. Why not better utilize all the space and make the most relevant information the most prevalent. There is no point in mimicking that interface when you aren’t displaying the same information. Why not better utilize the areas highlighted in red in the photo?

[image]

philchfree, 2024-04-17:
I second this and it would be a departure (improvement) from the bambu offering on the X1. Why can't we have during printing all informations at our disposal, much like the device page of bambu studio rathen than clicking the below screen to access them. I would love to see when printing on the same screen, the fan speeds, the print speed and even an AMS and spool sketch to see which filament is printing. On the left instead of the panda, we can have a small picture of the printer until you can present the actual model which is printing.

kurim, 2024-04-17:
I agree with you, showing the current printing model like [ha-bambulab](https://github.com/greghesp/ha-bambulab) does already would be great and a very nice improvment.

In HA it looks like this:
[image]

Also bring in the AMS humidity index would be nice.

bigtreetech, 2024-08-01:
The real estate was reserved for coming features. All will be revealed in August.

closed 2024-08-01 by bigtreetech

gpambrozio, 2024-08-09:
Would also love this