bigtreetech/PandaTouch #55: Is leaving the screen on constantly while using DV power risky?

sanctas, 2024-03-29:
I'd prefer to have the screen on permanently while using DC power, but that gives me two worries: 

1. Constant strain on the battery. I want to use battery because I want to be able to quickly grab the Panda Touch off the mount without restarting it using battery. 

2. LED Burn-in. What's the risk here? Could you share what screen this device uses so I could check out the data sheet for it?

heelsmj, 2024-04-01:
There is a switch on the back of the device that disables the battery if you are going to be using constant DC power. You can also set it to time out and sleep the display. Did you read the manual?

sanctas, 2024-04-01:
Please actually read the question I asked.

heelsmj, 2024-04-01:
> Please actually read the question I asked.

I don't know why I bother trying to help people on the Internet.  You would think I would have learned my lesson by now.

sanctas, 2024-04-01:
If you had actually read my post (or maybe you did read it and simply lack the intelligence to understand it, either way I suppose) you would know that I already know there's a battery on the device, which is specifically what my question is about. You would also know that I already know there is a way to keep the screen on. 

My question, had you actually read it instead of trying to make a snarky comment, is if there are risks to leaving the battery in a state of perpetual charging while constantly drawing DC power and if LED burn in will happen. 

My question is directed at the engineers from BIQU that could provide information about their long term testing. My question was *not* for some random dude who can't read. 

In the future, try reading a post before replying, thanks.

heelsmj, 2024-04-01:
Ad hominem and straw man attacks, now this thread is complete! 

Perhaps you might try to not assume the worst of people. I was not being snarky; I was asking if you read the manual because this is explained in the manual. I don't always read the manual for an item I purchased. 

You already know there is a battery on the device? OK, great did I infer or directly say you did not? 

As far as the issue at hand. Of course, there is a risk of constantly charging the battery. That is precisely why they thoughtfully included a switch to disable the battery when using constant DC power. This is obvious due to the fact that they included the switch. I assumed you had not seen this, or otherwise, why would you be asking about
 something that is clearly explained? This is why I asked if you read the manual. But no, you immediately assume you know better and think I am trying to be snarky rather than maybe thinking I am genuinely trying to help. 

[image]

looxonline, 2024-04-01:
Guys...be cool. I am on holiday and don't have time (or the willpower) to read through this whole thread but I've been seeing some interesting email notifications coming through which suggest that it is less than friendly. Until tomorrow, I promise that whatever this is about is not worth having words over. I will clarify any questions on the touch tomorrow.

sanctas, 2024-04-01:
> Guys...be cool. I am on holiday and don't have time (or the willpower) to read through this whole thread but I've been seeing some interesting email notifications coming through which suggest that [...]

I went ahead and blocked him for being a troll. Hopefully that means he can't reply anymore, but I'm not sure how GitHub handles blocks since I've never had to do that before. 

My question is simply, will the battery degrade significantly and/or swell due to constantly having power draw on it while charging? And does the LED screen experience significant burn in if left on permanently?

Safemode, 2024-04-02:
> And does the LED screen experience significant burn in if left on permanently?

My unit had one of the faulty USB-C boards for DC5V power, so I had turned off the screen timeout in order to utilize the display, and it has burn-in from just a few days of being on. So I would definitely not leave the LED screen on permanently

sanctas, 2024-04-11:
> My unit had one of the faulty USB-C boards for DC5V power, so I had turned off the screen timeout in order to utilize the display, and it has burn-in from just a few days of being on. So I would [...]

Thanks for that info. A slightly transparent screensaver could maybe help with this? Not sure.

slothy89, 2024-04-19:
> I went ahead and blocked him for being a troll. Hopefully that means he can't reply anymore, but I'm not sure how GitHub handles blocks since I've never had to do that before.
>
> My question is [...]

Given how quick the PT reboots, what harm is there in pulling it off the dock, flicking to Battery, waiting 10 seconds to reboot and away you go?

What I'm not clear on, is if when on DC mode, does it charge the battery at all? eg, charge to a set amount then disconnect charge circuit, or is it totally bypassed? If it's a total bypass, my suggestion isn't ideal as you won't know the charge state when pulling it off.

andy-spoo, 2024-07-28:
> What I'm not clear on, is if when on DC mode, does it charge the battery at all? eg, charge to a set amount then disconnect charge circuit, or is it totally bypassed? If it's a total bypass, my suggestion isn't ideal as you won't know the charge state when pulling it off.

Good point. If it does not trickle charge the battery at all when on DC there is a chance that the battery will slowly self discharge and when a LIPO battery voltage gets too low it pretty much kills the battery (it will never charge again).

bigtreetech, 2024-07-28:
The TFT screen will most certainly experience burn in from being on all the time. All TFT screens will to some extent. Modern technologies like OLED are even worse.

The battery will always be charging as long as it is on the dock regardless of the mode it is set to. The mode only defines whether or not that charge is being sent to the rest of the circuit.

closed 2024-07-28 by bigtreetech