bigtreetech/PandaTouch #57: Printer Connection Failed

jonjmills, 2024-04-02:
Brand New Unit setup. Connected to same WIFI/Network as printer.

Scan for Printers produces 0 results.
Manual setup of printer ends in Failed to connect.

Printer Firmware: 01.05.02.00  Cloud mode
Panda Touch Firmware: 1.0.1

Unfi Network and AP points.

Any suggestions to test?

bigtreetech, 2024-04-02:
Very likely that your network is blocking intra device to device comms. This is the case with many "guest" type networks. There is often a setting on the router to enable comms between devices on your network.

jonjmills, 2024-04-03:
Did some more troubleshooting on this and there is def something on myside blocking.

Created new network and WIFI SSID and everything connected and worked without issue.
Tried again on the original network/SSID with all firewall rules turned off and settings matched to the new network/SSID, still no go.

I am at a loss on what could be blocking but its not the device, closing the issue.

closed 2024-04-03 by jonjmills

bigtreetech, 2024-04-03:
Glad to hear that it came right.

bigtreetech, 2024-04-03:
> Did some more troubleshooting on this and there is def something on myside blocking.
>
> Created new network and WIFI SSID and everything connected and worked without issue. Tried again on the original [...]

BTW, firewall rules apply to data coming in from WAN and going out to WAN but not intra. There will be a separate setting for that.