bigtreetech/PandaTouch #59: Add Always on during Print

maxxe88, 2024-04-04:
It would be great to have a setting where the display stays on during a print and then goes back to the set time afterward.

Slider99, 2024-04-11:
I dont think this is a good idea, so I am printing something that takes 3 days and the display would stay on? What about screen burn in?

maxxe88, 2024-04-11:
So firstly, the "Always On" function already exists, and secondly, you don't have to use it. There are surely people whose prints don't take as long, for example, me. Mine usually take between 3-5 hours on average.

Gubbfittan, 2024-05-31:
Or there could be a function for always on where the display flips between settings with a selectable duration. like a screensaver

andy-spoo, 2024-07-28:
I'm not sure if screen burn is actually an issue. That was a 'thing' for old fashioned screens.
Bigtreetech makes MANY after market screen upgrades for 3D printers and all of them keep the screen constantly on during the print.
There is no reason to think that the Bamboo Touch is using a different kind of screen technology.
The screen off timer is purely there to save battery life when used in portable mode.