bigtreetech/PandaTouch #60: Beta Firmwares should be release in github.

superceleron, 2024-04-04:
I think closing beta firmwares inside a closed group in facebook, is not a good idea.
it should be release in here, i really don't understand why its not.

philchfree, 2024-04-04:
BigTreeTech asked for the github account either on facebook, youtube or reddit so I think that's the plan

superceleron, 2024-04-04:
well it should be posted here(readme is there for that), many ppl don't use any of that platforms.

philchfree, 2024-04-04:
Oh OK, did not get it. It was on multiple platforms though

bigtreetech, 2024-04-05:
It was posted on all of our social media platforms with the intention of gathering beta users who have enough technical knowledge to be able to test the firmware by themselves and provide accurate feedback. Unfortunately, some users often feel that beta firmware is just an early release and it can be quite distracting to the beta program when they start to ask simple questions about the firmware so we limited those who could participate instead of making it open to everyone. Github was not included in the invitation list but perhaps If it reaches a point where we need to add more beta users to the program then we will take this channel into consideration too.

philchfree, 2024-04-05:
Does this mean all beta users have been contacted already? If I did not receive anything, does this mean I have not been selected or it can still come?