bigtreetech/PandaTouch #70: backlight setting and infinite rebooting

jlo1975, 2024-04-12:
the devise use the USB port of the P1S

When trying to lower the backlight, the device reboot and return to the original 100% setting.

switching to battery seem to resolve the issue (i managed to set around 30-40 %)
BUT
when switching back to USB port the devise enter in endless rebooting 
the screen flicker every 1-2 sec with no logo showing and is no more responsive .....

I can recover the devise by switching back to battery mode .
So I restored the light setting to 100% until a fix is found.

DoTsTeR77, 2024-04-12:
Mine does this as well.. as soon as I even touch that slider it reboots.
On beta firmware 1.03 I should add. I going to try to official release to
see if it still occurs..

dilixi, 2024-04-15:
> the devise use the USB port of the P1S
>
> When trying to lower the backlight, the device reboot and return to the original 100% setting.
>
> switching to battery seem to resolve the issue (i managed to [...]

Is there a problem connecting to Panda Touch through TypeC using a mobile phone charger

jlo1975, 2024-04-16:
> Is there a problem connecting to Panda Touch through TypeC using a mobile phone charger

tested with 2 charger and got the same result when touching the backlight setting slider but more brutal than with the  P1S USB source. 
By more brutral I mean -> direct shutdown of the panda touch without infintie rebooting acting like a bricked device.
I've to plug out the cable from the charger and wait a few second then replug to get back the panda touch ok.

dilixi, 2024-04-17:
> > Is there a problem connecting to Panda Touch through TypeC using a mobile phone charger
> 
> tested with 2 charger and got the same result when touching the backlight setting slider but more brutal than with the P1S USB source. By more brutral I mean -> direct shutdown of the panda touch without the infinite rebooting. Acting like a bricked device. I've to plug out the cable from the charger and wait a few second then replug to get back the panda touch ok.

The backlight control function works well, possibly due to hardware issues.

andy-spoo, 2024-05-09:
If it's USB charger related then try using a basic charger with a charge only (as in no data) cable. 
It could possibly be an issue with quick chargers that use data to negotiate a charging speed/specification.

jlo1975, 2024-05-09:
> If it's USB charger related then try using a basic charger with a charge only (as in no data) cable. It could possibly be an issue with quick chargers that use data to negotiate a charging speed/specification.

hello
no it is not related to the usb charger as it is the same problem when using the USB port of the P1S.
afeter some search on internet it seems the first batch of panda touch have a very sensitive voltage controler that reboot the device on minor voltage fluctuations. It seems bigtreetech has release a B version of the little usb circuit that prevent the problems.

So I've openend a ticket to get this new usb board.

more info here:
https://www.reddit.com/r/BIGTREETECH/comments/1au5agu/panda_touch_problems/
https://www.reddit.com/r/BIGTREETECH/comments/1b17zux/updates_on_the_panda_touch_please_read_the/
and a photo of the new usb board (seems like they added some condos)
https://www.reddit.com/r/BIGTREETECH/comments/1bp8ozj/panda_touch_problems_updated/

jlo1975, 2024-07-16:
for info, I got no issue after getting the B version of the little usb circuit board.

bigtreetech, 2024-07-17:
> for info, I got no issue after getting the B version of the little usb circuit board(having a little voltage regulator chip) So the problem is hardware related for the first batches of panda touch.

Got you. Thanks so much for the update. Closing this out for now as it is not software related.

closed 2024-07-17 by bigtreetech