bigtreetech/PandaTouch #75: Image converter for Linux / macOS

andypiper, 2024-04-19:
Can we get the image conversion tool for Linux and macOS - would it be possible to release the source code so that we can compile it ourselves?

It looks like it is mostly a wrapper around lodepng but I'm not certain what the output format is so I can't reproduce it directly.

gpambrozio, 2024-08-10:
See [here](https://github.com/bigtreetech/PandaTouch/issues/108#issuecomment-2278942042)

dilixi, 2024-11-19:
> Can we get the image conversion tool for Linux and macOS - would it be possible to release the source code so that we can compile it ourselves?
> 
> It looks like it is mostly a wrapper around lodepng but I'm not certain what the output format is so I can't reproduce it directly.

@andypiper 

Hi,
there is [online tool](https://pandatouch.us.kg/) to change the panda image, and you can use it on local by this [rep](https://github.com/bigtreetech-beta/PandaTouch_Addtion_IMG).

And please close this issue thanks.

closed 2024-11-20 by BTTJUSTIN