bigtreetech/PandaTouch #79: AMS Humidity

ModeratorPS, 2024-05-02:
I would like it, if the Panda Touch is showing the AMS Humidity:

[image]

dilixi, 2024-11-19:
> I would like it, if the Panda Touch is showing the AMS Humidity:
> 
> ![IMG_7441](https://private-user-images.githubusercontent.com/102473254/327465923-1ab0800f-b959-4122-876d-354204f36a6f.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzIwMDgwMzAsIm5iZiI6MTczMjAwNzczMCwicGF0aCI6Ii8xMDI0NzMyNTQvMzI3NDY1OTIzLTFhYjA4MDBmLWI5NTktNDEyMi04NzZkLTM1NDIwNGYzNmE2Zi5qcGc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQxMTE5JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MTExOVQwOTE1MzBaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT02NWIzOTE0MDQwZWQ1NzY4ZDllZmJlZGExOWZiYWNkZThhNDBhYTg5YTRiOGZkYjMxNGM4N2Q2ZjU1ZjlmZDE4JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.4DC52GgpjNaAf2DTVz211JyYhp9vyxCL5qgWzxOvts4)

@ModeratorPS 

Hi,
This was implemented on V1.0.6,Please close this issue,thanks.

https://private-user-images.githubusercontent.com/56668660/387573390-442bb4e5-a8d1-4828-905b-52cfa9c08d22.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzIwMDgwNTMsIm5iZiI6MTczMjAwNzc1MywicGF0aCI6Ii81NjY2ODY2MC8zODc1NzMzOTAtNDQyYmI0ZTUtYThkMS00ODI4LTkwNWItNTJjZmE5YzA4ZDIyLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDExMTklMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQxMTE5VDA5MTU1M1omWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWUwM2RjNjdhMTA5OTk5NWIxODAwZTZkNTQ5MGIzZmFiMjNlYWVhMWEwM2RlN2M2ZmNlNjM0YmI1OTRjNjk4MmYmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.ljgjLeUDkgQI2JrkVQ0_Wi3CDC-Jw4IWtAGmZaD4Q24

BTTJUSTIN, 2024-11-20:
This bug was fixed in FW version V1.0.6

closed 2024-11-20 by BTTJUSTIN