bigtreetech/PandaTouch #80: Z-axis buttons reverted

r0mar0ma, 2024-05-13:
Buttons which are moving a plate in Z-axis are working in reverted direction.
Top buttons are moving the plate to the bottom, and bottom buttons are moving the plate to the top.
And also they have incomprehensible icons.

Firmware 1.0.3

philchfree, 2024-05-13:
Already closed https://github.com/bigtreetech/PandaTouch/issues/7

Problem for BTT is the panda touch can be used on A1 series and not only P1

LiamRay10, 2024-05-19:
> Already closed #7
> 
> Problem for BTT is the panda touch can be used on A1 series and not only P1

Then just make it printer to printer basis, example, "if printer=p1 or x1, then set z icons to classic, if printer=a1 or a1 mini, then set z icons to bedslinger"

LiamRay10, 2024-05-19:
and the z still moves upward on the a1 aswell in the same way you would visialize it.

philchfree, 2024-05-19:
> and the z still moves upward on the a1 aswell in the same way you would visialize it.

Yes but in this case it works, clicking UP moves the head up because that is the way Z is increased while on the P1/X1 to increase Z by clicking UP the bed goes down...

philchfree, 2024-05-19:
> > Already closed #7
> > Problem for BTT is the panda touch can be used on A1 series and not only P1
> 
> Then just make it printer to printer basis, example, "if printer=p1 or x1, then set z icons to classic, if printer=a1 or a1 mini, then set z icons to bedslinger"

That's what is suggested in this same thread https://github.com/bigtreetech/PandaTouch/issues/7

bigtreetech, 2024-07-23:
Inverting button order but not function in 1.0.5

BTTJUSTIN, 2024-09-13:
This feature was implemented in FW version V1.0.5

closed 2024-09-13 by BTTJUSTIN