bigtreetech/PandaTouch #81: Panda touch blocked latest bambu beta

insp1re1, 2024-05-17:
Panda touch blocked latest bambu beta firmware

kurim, 2024-05-18:
To be more clear:
[image]

The root cause is that only one MQTT Control channel can be used.
To resolve this problem PandaTouch should also support Cloudconnect.

philchfree, 2024-05-18:
Not sure what BTT can do if Bambu restrict MQTT commands when connected to cloud. It doesn't seem the issue can be resolved by BTT, I'm always on LAN mode so it does not affect me but I understand it can be a burden for people using cloud mode. Panda Touch is not broken by this new bambu firmaware restricts using both cloud and Panda Touch.

kurim, 2024-05-18:
But yes the reason is strange, I have HA running in Cloude mode & PandaTouch running w/o any issues.

Like the HomeAssistant addon it would be the best to offer a cloud connect.
Or like the NodeRed Setup mentioned here:
https://www.wolfwithsword.com/bambulab-to-home-assistant-nodered-configurator/

looxonline, 2024-05-18:
> Not sure what BTT can do if Bambu restrict MQTT commands when connected to cloud. It doesn't seem the issue can be resolved by BTT, I'm always on LAN mode so it does not affect me but I understand it can be a burden for people using cloud mode. Panda Touch is not broken by this new bambu firmware but is restricted that we can now not use both cloud and Panda Touch. I'm suspicious of the reason given by Bambu though...

We have never experienced a single instability with running local and cloud based MQTT and we have done a significant amount of testing. We have also never heard of anyone damaging their machine. The commands are no different to those that can be sent from the slicer or the handy app so there is an equal probability of something like this happening via those interfaces. Nevertheless, we simply don't have access to the volume of data that they have so we cannot comment on the veracity of the claims.

What we can say is that there is a software update for the Panda Touch coming very soon that will allow each printer to either connect via LAN mode or via Cloud mode. In this way, users can use the Panda Touch with a blend of machines. If Cloud mode access is limited then we have another access mechanism that we will employ.

insp1re1, 2024-05-18:
Lan mode and octo everywhere may provide same functionality?

superceleron, 2024-05-18:
Why i suspected this was coming... im using both for a long time never had a issue...  something smells fishy in bambu kingdom..

insp1re1, 2024-05-18:
I'm just glad it can be flashed to a klipper touch so I don't have a paper
weight.

insp1re1, 2024-05-18:
> GitHub <#81 (comment)
> As I stated, the next version of Panda Touch firmware will enable cloud
> MQTT and the Panda Touch will continue to work without a problem.

Can we jump on this absolute bad boy of a beta now?

looxonline, 2024-05-18:
> Can we jump on this absolute bad boy of a beta now?
> […](#)

Not just yet. It is working on the bench but we want to run it through a bit more testing before going out to the public. Probably another week.

LiamRay10, 2024-05-19:
> > Can we jump on this absolute bad boy of a beta now?
> > […](#)
> 
> Not just yet. It is working on the bench but we want to run it through a bit more testing before going out to the public. Probably another week.

Might be nice to also add the ability to change the button theme for issue #80 and allow for the button accent color to be changed from red to whatever the user so pleases.

insp1re1, 2024-05-19:
I think priority on it working first would be better

looxonline, 2024-05-20:
> I think priority on it working first would be better
> […](#)

Yes, we are making this priority and dedicating a firmware release to it alone. Other features are still being chipped away at in the background though and will come in a later release.

LiamRay10, 2024-05-23:
> > Can we jump on this absolute bad boy of a beta now?
> > […](#)
> 
> Not just yet. It is working on the bench but we want to run it through a bit more testing before going out to the public. Probably another week.

any update?

looxonline, 2024-05-24:
> > > Can we jump on this absolute bad boy of a beta now?
> > > […](#)
> > 
> > 
> > Not just yet. It is working on the bench but we want to run it through a bit more testing before going out to the public. Probably another week.
> 
> any update?

Working well. Literally in a meeting discussing UI refinements to make it as simple as possible for users to take advantage of. Beta will be out next week.

insp1re1, 2024-05-27:
Come on now

looxonline, 2024-05-27:
> Come on now
> […](#)

When it is ready, it will be released as a beta.

looxonline, 2024-05-27:
> Happy to take cash quickly though
> […](#)

Have you ever done any software development?

insp1re1, 2024-05-27:
Yeah

looxonline, 2024-05-27:
> Yeah
> […](#)

[image]

looxonline, 2024-05-27:
> Yeah
> […](#)

We will not release the firmware in an unfinished state. Our target is to have the first beta as a release candidate with most bugs already patched. That takes time and iteration.

insp1re1, 2024-05-27:
So what? That means I can't code ? 😂

insp1re1, 2024-05-27:
Looking at state of panda touch id say this is your first coding rodeo to
be honest

closed 2024-07-23 by bigtreetech