bigtreetech/PandaTouch #82: Add Translation

arolland, 2024-05-20:
Good morning,

Would it be possible to add additional languages ​​that can be modified by the community or by yourselves?

Thanks :)

looxonline, 2024-05-21:
Thanks for making this suggestion. We will discuss it internally and let you know if it will be scheduled for a future firmware release

arolland, 2024-05-21:
Thx :)

I can help you for trad (english to french), If you wan't :)

tadeu2, 2024-07-29:
I can help to translate to spanish. I had translated Orcaslicer entirely. Check it on official repo.

bigtreetech, 2024-09-16:
@tadeu2 @arolland and any others. We are planning multi language support in subsequent releases. Languages can be changed by uploading a new IMG file depending on the lang that the user desired to use. We can get Spanish and French files ready for the next release if you are happy to help with the translation. I am going to upload the yml to this repo shortly and then I will post the link here. If you go to that link, it will contain all of the instructions to do the translation. Thanks so much in advance for your help in getting the UI translated into different langs. Also, please use this thread if you have any questions about the meaning of the original English and we will help to clarify.

bigtreetech, 2024-09-16:
https://github.com/bigtreetech/PandaTouch/tree/master/Translation

tadeu2, 2024-09-17:
Spanish translation done.

ModeratorPS, 2024-09-21:
German translation done.

fabian39343, 2024-10-24:
Hi everyone,

are there step-by-step instructions on how to get a language file like this onto the Panda Touch?

I've read a lot already, but I can't figure it out at the moment.

best regards

bigtreetech, 2024-10-24:
Hey @fabian39343 they will all roll out with 1.0.6 which is dropping soon.

dilixi, 2024-11-19:
> Good morning,
> 
> Would it be possible to add additional languages ​​that can be modified by the community or by yourselves?
> 
> Thanks :)

@arolland

Hi,
there is [online tool](https://pandatouch.us.kg/) to change the panda image, and you can use it on local by this [rep](https://github.com/bigtreetech-beta/PandaTouch_Addtion_IMG).

And please close this issue thanks.

dilixi, 2024-11-19:
> > Good morning,
> > Would it be possible to add additional languages ​​that can be modified by the community or by yourselves?
> > Thanks :)
> 
> @arolland
> 
> Hi, there is [online tool](https://pandatouch.us.kg/) to change the panda image and the label on Panda Touch UI, and you can use it on local by this [rep](https://github.com/bigtreetech-beta/PandaTouch_Addtion_IMG).
> 
> And please close this issue thanks.

miniskull14, 2024-11-20:
English to French done, 
i have a beginner level
i am listening to you for the slightest question or suggestions

Zolexke, 2024-12-07:
[du-BE.zip](https://github.com/user-attachments/files/18046605/du-BE.zip)
Dutch translation done.

tadeu2, 2024-12-07:
You could do a pull request.

Spsnaky, 2024-12-22:
> Hey@fabian39343Sie werden alle mit der Version 1.0.6 eingeführt, die bald erscheint.

Ich habe nun version 1.0.6.2a drauf aber nur Englisch chinesich und Custom .
Wie kann ich es updaten auf Deutsch?

dilixi, 2024-12-22:
> > Hey@fabian39343Sie werden alle mit der Version 1.0.6 eingeführt, die bald erscheint.
> > @bigtreetech
> 
> Ich habe nun version 1.0.6.2a drauf aber nur Englisch chinesich und Custom . Wie kann ich es updaten auf Deutsch?

> > Hey@fabian39343Sie werden alle mit der Version 1.0.6 eingeführt, die bald erscheint.
> > @bigtreetech
> 
> Ich habe nun version 1.0.6.2a drauf aber nur Englisch chinesich und Custom . Wie kann ich es updaten auf Deutsch?

Hi,
You can use this.img file and change the language to Custom.

https://github.com/bigtreetech/PandaTouch/blob/master/Firmware/1.0.6.2/fat_v1.0.6_de-DE.img

Spsnaky, 2025-01-01:
> > > Hey@fabian39343Sie werden alle mit der Version 1.0.6 eingeführt, die bald erscheint.
> > > @bigtreetech
> > 
> > 
> > Ich habe nun Version 1.0.6.2a drauf, aber nur Englisch Chinesisch und Custom . Wie kann ich es auf Deutsch aktualisieren?
> 
> > > Hey@fabian39343Sie werden alle mit der Version 1.0.6 eingeführt, die bald erscheint.
> > > @bigtreetech
> > 
> > 
> > Ich habe nun Version 1.0.6.2a drauf, aber nur Englisch Chinesisch und Custom . Wie kann ich es auf Deutsch aktualisieren?
> 
> Hallo, Sie können diese IMG-Datei verwenden und die Sprache auf Benutzerdefiniert ändern.
> 
> https://github.com/bigtreetech/PandaTouch/blob/master/Firmware/1.0.6.2/fat_v1.0.6_de-DE.img

Vielen Dank hat funktioniert

Philippe26150, 2025-01-26:
> Salut [@fabian39343](https://github.com/fabian39343)ils seront tous déployés avec 1.0.6 qui va bientôt tomber.

Bonjour, 

J'ai passé mon Panda touch en version 1.0.6.3 et je ne vois pas la langue française. Il n'y a que English, Chinois ou Custom. 

Comment passer le Panda Touch en Français.

Merci

Uzuniita, 2025-01-29:
hi, there is someone can help me to have an Italian translation? it would be cool and usefull

Winky1st, 2025-02-01:
Hi, this .rar file contains my contribution for a full italian V 1.0.6.3 translation yml and img.
cheers all.

[Italiano - 1.0.6.x.zip](https://github.com/user-attachments/files/18630027/Italiano.-.1.0.6.x.zip)

Winky1st, 2025-02-01:
> hi, there is someone can help me to have an Italian translation? it would be cool and usefull

Allegato di seguito! Enjoy

Uzuniita, 2025-02-01:
> > hi, there is someone can help me to have an Italian translation? it would be cool and usefull
> 
> Allegato di seguito! Enjoy

Grazieeeee sei un grande! Mi ricordi come caricarlo per favore? Cheeers

Winky1st, 2025-02-01:
> > > hi, there is someone can help me to have an Italian translation? it would be cool and usefull
> > 
> > 
> > Allegato di seguito! Enjoy
> 
> Grazieeeee sei un grande! Mi ricordi come caricarlo per favore? Cheeers

Browse your panda touch IP and on BTT PANDA-TOUCH SETTINGS MANAGER load the img file. Enjoy

Uzuniita, 2025-02-01:
> > > > hi, there is someone can help me to have an Italian translation? it would be cool and usefull
> > > 
> > > 
> > > Allegato di seguito! Enjoy
> > 
> > 
> > Grazieeeee sei un grande! Mi ricordi come caricarlo per favore? Cheeers
> 
> Browse your panda touch IP and on BTT PANDA-TOUCH SETTINGS MANAGER load the img file. Enjoy

PERFECT worked and the translate its gooooood! thank you a lot!

Winky1st, 2025-02-01:
> PERFECT worked and the translate its gooooood! thank you a lot!

If you haven't done yet, reload the zip file end re-flash your Panda Touch . On the first upload there was 1 typo I've corrected. have fun. ciao!

Uzuniita, 2025-02-01:
I didnt see any error, where is It? Btw i re-flash It tomorrow! Thanks a LOT !

Winky1st, 2025-02-02:
> I didnt see any error, where is It? Btw i re-flash It tomorrow! Thanks a LOT !

Please, if you find any typo or words too long for the specific field, please do not hesitate to quote me here I will revise it - if any..
cheers

Winky1st, 2025-02-03:
I apologize if I had to re-upload an updated/revised file, but some too long words were messing up things.
 
- Fixed.

Please, if you notice any further errors, do not hesitate to point them out to me.

Enjoy - cheers

[Italiano - 1.0.6.x.zip](https://github.com/user-attachments/files/18635403/Italiano.-.1.0.6.x.zip)

Winky1st, 2025-02-03:
> I didnt see any error, where is It? Btw i re-flash It tomorrow! Thanks a LOT !

Unfortunately there were ...and not just one. As you know in Italian the words are very long sometimes and in some windows they were messing up things . I have updated the file again, I suggest you download the revised and corrected version. 
I apologize for this - 
wish you a nice day.

hitroz, 2025-02-10:
Не магу перевести на русский язык кто может помочь мне с переводом

hitroz, 2025-02-11:
Можете пожалуйста добавить русский язык  с переводом засада помогите пожалуйста мне .

zhekArcher, 2025-02-24:
почему не работает конвектор с yml в IMG  ????

rolan8881, 2025-03-02:
Помогите с переводом на русский язык .

dilixi, 2025-03-22:
We can use this tool to modify language text from panda touch.

https://github.com/bigtreetech-beta/PandaTouch_Addtion_IMG

closed 2025-03-22 by dilixi

Winky1st, 2025-03-26:
[it-IT_V1.0.7.zip](https://github.com/user-attachments/files/19471354/it-IT_V1.0.7.zip)

Italian V1.0.7 translation

Enjoy

Winky1st, 2025-04-26:
@dilixi hi bro and good day.
Can you please update https://generate-addtion-img.vercel.app/ to 1.0.7.1 fw ? I found 1 little typo on 1.0.7 Ita translation that I  wish to correct; I will upload the corrected file after it.
Hint: why don't you add a third input field where will be possible to load the desired fw version? Doing so, in future  you won't  have to update it every time and the user can do it .

Thankyou - cheers

dilixi, 2025-04-26:
> [@dilixi](https://github.com/dilixi) hi bro and good day. Can you please update https://generate-addtion-img.vercel.app/ to 1.0.7.1 fw ? I found 1 little typo on 1.0.7 Ita translation that I wish to [...]

Hi,
"add a third input field"?
What do you mean?

I will check the https://generate-addtion-img.vercel.app/ next week.

Winky1st, 2025-04-26:
> Hi, "add a third input field"? What do you mean?
>
> I will check the https://generate-addtion-img.vercel.app/ next week.

Just an idea and I don't know if it is possible. but why should you upgrade FW every time inside that useful tool? (happened the same from 1.0.6 to 1.0.7 in past) so the idea was, why not to add a third input field where user can drop in the desired FW_xxx.bin downloaded from your github + yml file with translation + optional png file, In this case the tool wont need your further actions to be updated to the new fw version. I hope i could explain what I meant. 
Cheers.

In any case,  the file below attached, is typo corrected (line 331) - ita yml file:

[it-IT.zip](https://github.com/user-attachments/files/19922981/it-IT.zip)

dilixi, 2025-04-29:
> > > [@dilixi](https://github.com/dilixi) hi bro and good day. Can you please update https://generate-addtion-img.vercel.app/ to 1.0.7.1 fw ? I found 1 little typo on 1.0.7 Ita translation that I wish to correct; I will upload the corrected file after it. Hint: why don't you add a third input field where will be possible to load the desired fw version? Doing so, in future you won't have to update it every time and the user can do it .
> > > Thankyou - cheers
> > 
> > 
> > Hi, "add a third input field"? What do you mean?
> > I will check the https://generate-addtion-img.vercel.app/ next week.
> 
> Just an idea and I don't know if it is possible. but why should you manually upgrade FW version every time inside that useful tool? (happened the same from 1.0.6 to 1.0.7 in past) so the idea was, why not to add a third input field where user can drop in the desired FW_xxx.bin downloaded from your github (usually the latest) + yml file with translation + optional png file, In this case the tool won't need your further actions to be updated to the new fw version. I hope I've been able to explain to you as best I can what I mean. Cheers.
> 
> In any case, the file below attached, is typo corrected (line 331) - ita yml file:
> 
> [it-IT.zip](https://github.com/user-attachments/files/19922981/it-IT.zip)

@Winky1st 

Okay.
I got you.
That's a good idea and I will do it in someday.

Thank you very much.

labsik08, 2025-05-19:
Hello, do you have the French translation of Panda Touch?
Have a nice day.

Winky1st, 2025-06-04:
@dilixi Hallo my friend and nice day. 
If you post eng.yml file for 1.0.7.3 I will add the new items translated.
cheers.

Philippe26150, 2025-06-05:
Hello, is there a French version? If so, how can I get it?
Sincerely

Bonjour, y a t il une version en français ? Si oui comment l'obtenir ?
Cordialement

dilixi, 2025-06-06:
> [@dilixi](https://github.com/dilixi) Hallo my friend and nice day. If you post en-en.yml file for 1.0.7.3 I will add the new items translated into it-it.yml cheers.

Hi,
I will do it but there are lot of things I need to do this month.

obadaw007, 2026-09-01:
Witam 
Czy jest możliwość spolczenia tego oprogramowania w jezyku Polskim 

Hello,
Is it possible to make this software available in Polish?