bigtreetech/PandaTouch #83: Revert files order in printer tab

r0mar0ma, 2024-05-29:
Currently files in printer tab are showing in forward order. But reverse order is much preferable, because in most cases it needs to print last printed file. Currently it requires really a lot of time to scroll down if you have tens of files on SD card.
And actually native printer interface is showing files in reverse order.

Also it will be helpful to sort them by date (in reverse order), not by the order how they are placed in FAT structure on the card. But not sure do the device have sufficient hardware resources for that...

BTTJUSTIN, 2024-06-14:
Thanks for making this suggestion. We will discuss it internally and let you know if it will be scheduled for a future firmware release

dilixi, 2024-11-19:
> Currently files in printer tab are showing in forward order. But reverse order is much preferable, because in most cases it needs to print last printed file. Currently it requires really a lot of [...]

@r0mar0ma 

Hi,
This was implemented on V1.0.5,Please close this issue,thanks.

BTTJUSTIN, 2024-11-20:
This bug was fixed in FW version V1.0.5

closed 2024-11-20 by BTTJUSTIN