bigtreetech/PandaTouch #84: 3rd party light controls

Foroxon, 2024-06-01:
I have an additional LED strip to fix P1S's poor factory light issue.
It would be cool to have the ability to control that light from the Panda Touch screen.
I'm using [WLED](https://github.com/Aircoookie/WLED) controller for my additional light.
Could you add some configurable webhook or something like that to control it?
Thanks!

scobby, 2024-06-12:
I use home assistant for that case, you can sync the light entitiy from the printer to every other light in your home.

Foroxon, 2024-06-12:
> I use home assistant for that case, you can sync the light entitiy from the printer to every other light in your home.

Yes, I did the same HA automation for 3rd party light. But there is a 2-3 seconds latency between factory and 3rd party light.

BTTJUSTIN, 2024-07-10:
Hi, the 3rd party light controls function will be enabled in our new product ,please pay attention to our new product releases

closed 2024-07-26 by BTTJUSTIN