bigtreetech/PandaTouch #87: BambuLab released new firmware for P1 series that will break support on Panda touch unless on lan mode

zeemyself, 2024-06-11:
https://wiki.bambulab.com/en/p1/manual/p1p-firmware-release-history

at the bottom of release notes

Please note that this restriction will affect the use of software and hardware such as Home Assistant and BiQu Panda Touch. Suppose you still need to use such software and hardware to control the restricted functions. In that case, we recommend enabling the printer's LAN Only mode to continue using the restricted control functions with the Local MQTT Broker.
We will detail these restrictions in the firmware update notes to ensure users are informed in advance.

superceleron, 2024-06-11:
its already reported here: https://github.com/bigtreetech/PandaTouch/issues/81

Also im on the latest beta (1.05.04.12) and its working fine again with the pandatouch.

superceleron, 2024-06-11:
yes im sure, im using their app and pandatouch at the same time just fine,

insp1re1, 2024-06-11:
can confirm it works with panda, and now shows files in the sd card on the panda, it didnt before.

looxonline, 2024-06-11:
It will work with the new bambu firmware but starting prints, heating the extruder and heating the bed will be disabled when the printer is in cloud mode. I am working with the new BIGTREETECH cloud mode firmware at the moment on my machines. There are a few kinks that still need to be ironed out but we are now very close to a beta.

superceleron, 2024-06-11:
Ye i notice that now is only partially working, im i bit busy and not using much this days the printer.

insp1re1, 2024-06-11:
Disgrace

zeemyself, 2024-06-13:
Can we have a list of what works and what doesn't compared to the previous version?

insp1re1, 2024-06-13:
Try it

insp1re1, 2024-06-13:
Try it bro

looxonline, 2024-06-13:
> Can we have a list of what works and what doesn't compared to the previous version?

Check my previous comment for that list. As for the new beta that we will be releasing, everything works again. The beta still needs some love but I guess that is why it is a beta ;)

looxonline, 2024-06-13:
> > > Can we have a list of what works and what doesn't compared to the previous version?
> > 
> > 
> > Check my previous comment for that list. As for the new beta that we will be releasing, everything works again. The beta still needs some love but I guess that is why it is a beta ;)
> 
> Thank you, also is it better to have a detailed description in the [wiki](https://bttwiki.com/PandaTouch.html) instead of this Github issue

True. We are busy working on some comms at the moment and will add a wiki update to the list.

bl4ckm0onk3y, 2024-06-14:
Confirmed on the new 1.06 firmware on P1P@P1S (all hardware upgrades) the 1.03 Panda Touch in Cloud Mode some features do not work, like starting prints, controlling the temperatures.

philchfree, 2024-06-15:
Seems FW1.04 is out but how do we access it? I asked 3 months ago to be part of the beta revo but I've never ever received acknowledgment! Can you please add me (again?)?

insp1re1, 2024-06-15:
They was supposed to add everyone but never did

looxonline, 2024-06-15:
> Seems FW1.04 is out but how do we access it? I asked 3 months ago to be part of the beta revo but I've never ever received acknowledgment! Can you please add me (again?)?

The signup has been sent out via email, twitter and FB. Here it is again for completeness. Users will be bulk added on Monday when the team is back in the office. https://forms.gle/m6FGCDW2yy2dJ538A

awganponton, 2024-06-15:
Hello all. Is it just me or all of us. I can change temp of the nozzle in panda touch. 
You can set it but it does not do anything.

looxonline, 2024-06-16:
> Hello all. Is it just me or all of us. I can change temp of the nozzle in panda touch. You can set it but it does not do anything.

Please read through the entirety of the thread. All the answers are in here.

awganponton, 2024-06-16:
Actually your reply... I've missed that. Thanks.

Foroxon, 2024-06-17:
> The signup has been sent out via email, twitter and FB. Here it is again for completeness. Users will be bulk added on Monday when the team is back in the office. https://forms.gle/m6FGCDW2yy2dJ538A

Requested access to the beta program a few days ago but didn't get any email with further instructions today. Should I wait more, or do I need to submit some other form?

looxonline, 2024-06-17:
> Requested access to the beta program a few days ago but didn't get any email with further instructions today. Should I wait more, or do I need to submit some other form?

Those who signed up before Thursday last week should have received access today. For those after Thursday last week, they will receive access tomorrow (I'm only back at work tomorrow as it was a national holiday in South Africa today). The email address used to sign up to github will get an email letting them know that they have access to the repo.

LivinDuck, 2024-06-19:
I just signed up for the beta repo, when might I be getting access?

insane78, 2024-06-20:
> I just signed up for the beta repo, when might I be getting access?

same here

cayalag, 2024-06-20:
Good to hear the beta is out !! I just signed up as well :)

looxonline, 2024-06-20:
> I just signed up for the beta repo, when might I be getting access?

Will do another batch add tomorrow.

beastyboyuk, 2024-06-21:
Signed up 3 days ago still no reply OMG

RobertoMacchi, 2024-06-21:
Good morning everyone and excuse my ignorance, I received the Beta testing invitation and I would like to know if by installing the new FW 1.04beta on the Panda Touch I can currently overcome the Cloud blocks of the new firmware 1.06 Bambu Lab for P1x, especially the one that no longer allows me to give the command to print files directly from the list displayed on Panda Touch, but forces me to use the uncomfortable and stupid original display of P1x. Thank you for your patience and your answers.

looxonline, 2024-06-21:
> Signed up 3 days ago still no reply OMG

Yea, users are added in batches once a week. See my previous replies.

looxonline, 2024-06-21:
> Good morning everyone and excuse my ignorance, I received the Beta testing invitation and I would like to know if by installing the new FW 1.04beta on the Panda Touch I can currently overcome the [...]

Yes, the 1.0.4 beta will overcome these blocks.

awganponton, 2024-06-21:
For me after change wifi started to be super unreliable. When i will aproach printer every time its lost...
And from router to printer i have 3 meters of clear air ...
It was never good because from time to time connection was lost but now it is worse than it was.

looxonline, 2024-06-21:
> For me after change wifi started to be super unreliable. When i will aproach printer every time its lost... And from router to printer i have 3 meters of clear air ... It was never good because from time to time connection was lost but now it is worse than it was.

Thanks for the feedback but please log any issues with the beta, in the beta repo. We have different teams monitoring each repo and your feedback will make it to the right team in the fastest way if you record it in the correct repo.

beastyboyuk, 2024-06-21:
> > Signed up 3 days ago still no reply OMG
> 
> Yea, users are added in batches once a week. See my previous replies.

I understand but only had my panda 2 days upgraded and now its a brick ? . Also isnt the real firmware due very soon ? I just want to use my brick thats all

insp1re1, 2024-06-21:
Unlucky, works for rest of us, must be user error

beastyboyuk, 2024-06-22:
> Unlucky, works for rest of us, must be user error
> […](#)

cheers for that great reply your a real nice guy

cayalag, 2024-06-22:
> > I just signed up for the beta repo, when might I be getting access?
> 
> Will do another batch add tomorrow.

Was that done in the end ? 
Thanks for your continued effort ;)

philchfree, 2024-06-22:
> > > Signed up 3 days ago still no reply OMG
> > 
> > 
> > Yea, users are added in batches once a week. See my previous replies.
> 
> I understand but only had my panda 2 days upgraded and now its a brick ? . Also isnt the real firmware due very soon ? I just want to use my brick thats all

Have you tried flashing back to FW1.03?

RobertoMacchi, 2024-06-22:
Good morning everyone, I have now updated my five Panda Touch installed on as many p1p/s to FW1.04beta4 and in addition to solving the Bambu Blocks problem, I have noticed a better, more stable and faster WiFi connection from the Panda Touch to the printer. I am not currently noticing any problems and I am very satisfied with the operation. I would therefore like to sincerely thank the authors who have worked hard to quickly resolve the problems introduced by version 1.06 of the Bambu Lab firmware. I would also like to know if, having also introduced the Bambu Lab account Login system in the new FW1.04 version, this can be considered a definitive override and therefore we can expect to have definitively resolved the Blocks problem also on subsequent versions of the Bambu Lab updates, or if the next BL versions may still reserve us some unpleasant surprises. Thanks for your answers and your courtesy.

LivinDuck, 2024-06-22:
Hey, any updates on when new signups may be added to the repo?

beastyboyuk, 2024-06-22:
> Have you tried flashing back to FW1.03?

My panda is on FW 1.03 I need the beta has my printer p1s upgrade to the latest version and now my panda does not work lol it's been 5 days now ?

philchfree, 2024-06-23:
OK so downgrade your printer

beastyboyuk, 2024-06-23:
> OK so downgrade your printer

Lol did not no you can do that pmsl thanks but still i would like the beta firmware and not to wait a week the real firmware be out then

insp1re1, 2024-06-23:
Don't think you can handle it bro

looxonline, 2024-06-23:
> Hey, any updates on when new signups may be added to the repo?

Anyone who signed up between last Friday last week and Thursday this week should have been added on Friday of this week. Next batch add will be on Wednesday next week.

looxonline, 2024-06-23:
> Good morning everyone, I have now updated my five Panda Touch installed on as many p1p/s to FW1.04beta4 and in addition to solving the Bambu Blocks problem, I have noticed a better, more stable and [...]

This entirely depends on Bambu Lab. They stated that they were concerned about commands coming from two sources and we have resolved that now. Commands can only come from the cloud.

cayalag, 2024-06-23:
> > Hey, any updates on when new signups may be added to the repo?
> 
> Anyone who signed up between last Friday last week and Thursday this week should have been added on Friday of this week. Next batch add will be on Wednesday next week.

I signed up last Thursday and haven't been added yet

looxonline, 2024-06-23:
> > > Hey, any updates on when new signups may be added to the repo?
> > 
> > 
> > Anyone who signed up between last Friday last week and Thursday this week should have been added on Friday of this week. Next batch add will be on Wednesday next week.
> 
> I signed up last Thursday and haven't been added yet

Thursday the 20th or the 13th?

cayalag, 2024-06-23:
The 20th. 4 days ago

bigtreetech, 2024-06-23:
> The 20th. 4 days ago>

Should be added on Monday.

beastyboyuk, 2024-06-23:
> > Hey, any updates on when new signups may be added to the repo?
> 
> Anyone who signed up between last Friday last week and Thursday this week should have been added on Friday of this week. Next batch add will be on Wednesday next week.

This is not true a lie i did it last tuesday and not heard nothing i dont understand why there is not a link for anyone who wants it has this system does not work i have NOT BEEN ADDED I HAVE CHECKED MY FORM AND EVERYTHING IS CORRECT

RobertoMacchi, 2024-06-23:
Good morning everyone, I am using firmware 1.04 version 4 on five Panda Touch connected to as many p1p/s printers. This morning I updated to the latest Panda Touch and strangely I cannot connect to my bambulab account. The wheel continues to spin for about 60 seconds and then gives me the warning "Failed to connect to my account". I performed the identical firmware update procedure on all five panda touch in my possession. I tried reinstalling the firmware but without success. I also tried changing the password of my bambulab account always with no positive results. The other four displays continue to work correctly, but I have not tried to log out and back in from my account on these four working devices because I fear I will also lose these connections. Does anyone have an idea of ​​what may have happened and possibly how to fix it? Thanks for your attention and for your answers.

bigtreetech, 2024-06-23:
> This is not true a lie i did it last tuesday and not heard nothing i dont understand why there is not a link for anyone who wants it has this system does not work i have NOT BEEN ADDED I HAVE [...]

Because the beta program is not for everyone. It is firmware that is still in development and we don't want users unwittingly downloading it and finding bugs in it without a full understanding that it is in development. If you have not been added already then you will be added as soon as the team gets to it. The beta program is already quite full and the team is fielding a lot of feedback. They are prioritising that feedback and bug fixes over new member additions so those who were not added on Friday will be added on Monday. Although there is no limit to the number of people we will allow as part of the public beta, we have finite hours in the day and must focus them on getting the firmware to release stage.

bigtreetech, 2024-06-23:
> Good morning everyone, I am using firmware 1.04 version 4 on five Panda Touch connected to as many p1p/s printers. This morning I updated to the latest Panda Touch and strangely I cannot connect to [...]

Thanks for your feedback. Please post beta feedback in the beta repo. This repo is for release feedback and beta feedback is not monitored here.

beastyboyuk, 2024-06-23:
> Because the beta program is not for everyone. It is firmware that is still in development and we don't want users unwittingly downloading it and finding bugs in it without a full understanding that [...]

Thanks for a reply at least . Well when you just bought the product and it does not work anymore i would be glad to use it :) better that then a brick

insp1re1, 2024-06-23:
Please stop posting shit all the time

bigtreetech, 2024-06-23:
> Please stop posting s*** all the time
 Blocked

bigtreetech, 2024-06-23:
> Thanks for a reply at least . Well when you just bought the product and it does not work anymore i would be glad to use it :) better that then a brick

We fully get you. Fortunately it still maintains a lot of the original functionality even when not using the beta. It is just the heating and print start that was blocked. I am sure that the team will wrap up adding users from the last batch on Monday and you will be back in business.

beastyboyuk, 2024-06-23:
> We fully get you. Fortunately it still maintains a lot of the original functionality even when not using the beta. It is just the heating and print start that was blocked. I am sure that the team [...]

Thanks very much

RemindZ, 2024-07-05:
@bigtreetech 
I believe I've requested access to the Beta repo twice now, once yesterday and once some weeks ago. So far I haven't received any notification / email letting me know that I can access it. 
Where would I be able to access it, in case I have already been added?

rtcmay, 2024-07-07:
> @bigtreetech I believe I've requested access to the Beta repo twice now, once yesterday and once some weeks ago. So far I haven't received any notification / email letting me know that I can access it. Where would I be able to access it, in case I have already been added?

me too

heelsmj, 2024-07-07:
I filled out the form to be added to beta weeks ago and have still not been added. Can you add me please?

poli27, 2024-07-08:
Please add  your nick to beta firmware thx

[5 more replies]

closed 2024-07-19 by looxonline