bigtreetech/PandaTouch #88: Custom Filament Profiles

ModeratorPS, 2024-06-22:
It would be nice, if you could select the custom Filament Profiles you have created

[image][image]
[image]

kurim, 2024-06-24:
Way to get is using the API for it, I mentioned it here for HA Addon:
https://github.com/greghesp/ha-bambulab/issues/540#issuecomment-2186383233

BTTJUSTIN, 2024-06-26:
Thanks for making this suggestion. We will discuss it internally and let you know if it will be scheduled for a future firmware release

t3chguy, 2024-07-10:
I guess the issue would be the marketing says it works entirely with no internet connection so BambuCloud API access would not be included

[image]

BTTJUSTIN, 2024-07-10:
Hi,This requirement requires high hardware resources, which are currently not available with PandaTouch. We are evaluating new solutions.

zimivanende, 2024-11-20:
I would like to reopen this discussion. Custom Filaments is an important feature. What is the use of the touch if you can't set the filament and still have to go in the slicer or app to set these. Bought this device so I can manage ALL from this touchscreen. If I have to go to the slicer to set the filement I can load, change settings, set temperatures from there as well. 

If you can't set it and decide not to implement it at least take away the limitation that you need to set a filament before being able to load or unload...

Side question why does the x1c have limit of 120 degrees C throught the panda touch and the P1S doesn't? When you add them in a group and set a P1S as master the x1c doesn't have this limit so it is possible...

YetAnotherBugHunter, 2024-11-28:
I second the request. I am having similar issues with custom filaments, to the point where the Panda Touch (PT) filament management page is non-functional.  Example, if I set a custom filament in the slicer or the Bambu Handy app, I cannot go the PT and attempt to load or unload a filament. If I try the the PT indicates that the filament is unknown and offer to allow me to edit the filament definition. The ability to manage filaments at the printer with the PT is the feature that prompted me to purchase the PT.

Given that the latest PT firmware can connect to the cloud account, could it not be possible to query the custom filaments associated with the account?

looxonline, 2024-11-28:
> I second the request. I am having similar issues with custom filaments, to the point where the Panda Touch (PT) filament management page is non-functional. Example, if I set a custom filament in the [...]

We are working on this and plan to have greater support in V1.0.7. @dilixi

dilixi, 2025-04-24:
> It would be nice, if you could select the custom Filament Profiles you have created
> 

Hi,
It should can work now.

Please have a try:
https://github.com/bigtreetech-beta/PandaTouch-Support/blob/main/firmware/v1.0.7.2_beta1.bin