bigtreetech/PandaTouch #92: Group View UI options

K2Kevin, 2024-07-08:
Adding a few printers to a group and viewing them on the Home Screen is a great feature.  This view would really benefit from more UI viewing options.  The current "Grid" view only allows for viewing of two printers. In order to view other printers status, you must scroll right or left, not very fluid interface.

**Feature Request:**
Line item column view.  Each printing on a single line with minimal print status that can fit the screen (enable of Bambu Studio's Multi-Device support)
[image]

BTTJUSTIN, 2024-07-10:
Thanks for making this suggestion. 
We will discuss it internally and let you know if it will be scheduled for a future firmware release.

K2Kevin, 2024-08-12:
> Thanks for making this suggestion. We will discuss it internally and let you know if it will be scheduled for a future firmware release.

Id say it's a must. The interface can be VERY slow (unusable) to scroll right to left when you are connected to 6 printers.  I use the PandaTouch to monitor multiple printers, and I can say with the current interface, its not very usable for this use case.

K2Kevin, 2024-08-12:
https://github.com/user-attachments/assets/1239e855-7c33-4ccd-8219-e1775b02f03c

K2Kevin, 2024-10-23:
> Thanks for making this suggestion. We will discuss it internally and let you know if it will be scheduled for a future firmware release.

Any new on this feature?  
I really want to love the Panda Touch, and use it to monitor multiple printers progress.  It's currently very difficult to monitor multiple printers due to the slow processor being used and interface.  This enhancement would solve this!  Simple list view could display 5-10 printers on one view without scrolling.

Also, the constant popups warning Connect Connect (because I turned off printer) needs to have an option to turn off this popup.

AverageHedgehog, 2024-10-27:
I've had a very similar experience and don't use my device much because of it

1. Slow/laggy interface
2. Home screen only shows 2 printers, where I want to list at least 4
3. Most of the time it's complaining that printers are disconnected, when some printers are turned off.
4. The home page shows a list of printers in order, I want the ability to re-order by print status (closest to finishing shows up first)

pezliz, 2024-10-27:
I was really hoping this would be a better way of managing my Bambus, but it's just so laggy and sending me notifications I don't need. I wish it was more intuitive and customizable.

bigtreetech, 2024-10-27:
We are working on a grid display and also planning to get rid of the swipe in favour of a paginated printer display. The swipe is what introduces the lag but pagination just changes from one grid to the next. Being able to sort printers in order of finishing first/finishing last is also a brilliant idea. @BTTJUSTIN @dilixi please take note of these for V1.0.7.

BTTJUSTIN, 2024-11-08:
OK，I will communicate with the PM about adding this feature to the V1.0.7.

dilixi, 2024-11-19:
> We are working on a grid display and also planning to get rid of the swipe in favour of a paginated printer display. The swipe is what introduces the lag but pagination just changes from one grid to [...]

we can do it after the UI work.

K2Kevin, 2025-01-31:
> OK，I will communicate with the PM about adding this feature to the V1.0.7.

I hope this will still be released, would bring life to this device for many of us!

K2Kevin, 2025-03-18:
ANY ETA on firmware 1.0.7?

K2Kevin, 2025-03-22:
> OK，I will communicate with the PM about adding this feature to the V1.0.7.

Looks like none of the features made into this release.  BUMMER!   Simple multi-device grid view PLEASE!

AverageHedgehog, 2025-04-14:
Any updates on this? I would love an option to toggle disconnected printers notifications off because it makes the unit unusable. I like to leave it on my desk and see what printers are printing, and check the status without touching anything. 6-8 printers on one screen would be ideal.

K2Kevin, 2025-05-29:
Still hope this happens,  we need a line item list of printers in group for easy view monitoring.

K2Kevin, 2025-08-14:
Hope this is still in the works.

K2Kevin, 2025-09-14:
Well...  It's been over a year since this request.  I guess we should assume it's just not coming to PandaTouch.  bummer...

bigtreetech, 2025-09-14:
There are development pipelines in place and with limited resources certain projects/features must get priority. Until this is either closed or implemented, it is not ruled out. As it stands, it is still open.

K2Kevin, 2026-02-03:
Of course when the Beta is released, I would love access!