bigtreetech/PandaTouch #95: Add option to turn off thumbnails/icons.

Vantage-DS, 2024-07-16:
V1.0.4Beta9

The thumbnails idea looks great, but with 100 files on my USB, it's now become impossible to find what i want to print.

The scrolling text and "Organise by newest" icons now mean i cannot find what i want to print without spending 5 minutes trying to read through the list.

looxonline, 2024-07-16:
Folders will be coming soon (hopefully 1.0.5) and then it should make things easier to nav.

Vantage-DS, 2024-07-16:
Would the 42 icon limit be applied to each folder?

I could handle that as it looks like a great idea 😀

bigtreetech, 2024-07-16:
> Would the 42 icon limit be applied to each folder?
> 
> I could handle that as it looks like a great idea 😀

I would need to chat to the dev who did this but it likely would.

We plan to get the file/folder browsing just right for each user but it will take a bit of time as each new addition requires a heap of testing to be sure that nothing else is broken. Such are the joys of software dev.

Vantage-DS, 2024-07-16:
Yeah it's coming along and I like the piece of kit a lot. Overall it's saved me a lot of time instead of slicing and sending everything each time (I refuse to use the Handy app/cloud mode). It's just what seems to work for me in a business environment doesn't seem to be how others enjoy using it 🙃

looxonline, 2024-07-17:
> Yeah it's coming along and I like the piece of kit a lot. Overall it's saved me a lot of time instead of slicing and sending everything each time (I refuse to use the Handy app/cloud mode). It's just what seems to work for me in a business environment doesn't seem to be how others enjoy using it 🙃

So we decided to pull thumbnails from 1.0.4 and include into 1.0.5 in a month from now. It will give us more time to refine it.

looxonline, 2024-07-17:
Keeping this open as a request for when we do 1.0.5.

dilixi, 2024-11-19:
> V1.0.4Beta9
> 
> The thumbnails idea looks great, but with 100 files on my USB, it's now become impossible to find what i want to print.
> 
> The scrolling text and "Organise by newest" icons now mean i cannot find what i want to print without spending 5 minutes trying to read through the list.

@Vantage-DS 

Hi,
We did add it on V1.0.5.
Please close this issue, thanks.

closed 2024-11-19 by Vantage-DS