bigtreetech/PandaTouch #98: Add FTP access to Panda Touch

philchfree, 2024-07-19:
Is there a way to access panda touch through FTP so that we can access the USB key remotely?

andy-spoo, 2024-07-28:
I suspect FTP has too many security issues leaving it vulnerable to attack =

1. Data sent in plain text: FTP transmits usernames, passwords, and data without encryption, making it vulnerable to eavesdropping, sniffing, and interception.
2. Weak authentication: FTP uses plain text passwords, which can be easily compromised through brute-force attacks or password cracking.
3. Lack of encryption: FTP does not provide built-in encryption for data transfer, leaving sensitive information exposed to tampering and interception.

philchfree, 2024-07-28:
FTP is available directly from the printer (port 990 (implicit ftp), user: bblp, password: access code) so wanted to have it also available when using panda touch to access the USB key.