bigtreetech/PandaTouch #100: Cannot update to V1.0,4 - Getting [HTTP ERROR] Bad Request

andy-spoo, 2024-07-28:
I'm trying to upate from firmware 1.0.3 to 1.0.4
When uploading the .bin file, on completion of the upload it says [HTTP ERROR] Bad Request so does not complete the update.
I did just before install the new Bamboo Update, version  01.06.01.00 which might not help (yes I know it might not be compatible but I updated before I was that 1.0.4 had come out.

Any ideas?

[image]

TMillesich, 2024-07-28:
The update of the BambuLab printer does not matter in this case. 
How long did it take to upload the panda_touch-v1.0.4.bin (1.76MB)? Should not be any longer than 3 seconds.

I'd try it multiple times, just to be sure that it's no WiFi hickup.

bigtreetech, 2024-07-28:
Could be that the wifi signal to your PC or the panda touch is bad. Try to move both to a better wifi area. An update requires great signal from both devices.

andy-spoo, 2024-07-30:
Both the laptop and Bamboo Touch were getting a good wifi signal but I have moved both closer to the router. The signal is very good and I'm still getting the same issue.
The file only takes about 2 seconds to upload but at the very end it still shows the same error.
I've tried using the Brave browser, Opera browser and Firefox browser which made no difference.
I've also tried the new V1.04.1 file with the same issue.
Is there any way to flash from USB storage unless anyone has any other options?

bigtreetech, 2024-07-30:
I wonder if there is not something on your home network that is blocking file transfer on port 80. Maybe try to set up a hotspot on your phone, connect your laptop and the PT and then try again.

andy-spoo, 2024-07-30:
> I wonder if there is not something on your home network that is blocking file transfer on port 80. Maybe try to set up a hotspot on your phone, connect your laptop and the PT and then try again.

I've just tried using my phone as a hotspot as you suggested. The result is the same. I've also just tried using Windows (I normally us Mint Linux) and it did not help.

bigtreetech, 2024-07-30:
@dilixi Please could you assist here. It seems like the file is being uploaded but at the end an HTTP error is being displayed. Maybe the user can program it via USB-C?

TechnoSwiss, 2024-07-30:
@andy-spoo I know Brave and Opera are both Chromium based, but have you given straight Chrome try? I was running into the same issue in Firefox and I tried running from Chrome on another machine and was able to run the update without issue. Maybe try running the browser in Private mode just to make sure there isn't something in the browser cache that's causing an issue.

bigtreetech, 2024-07-30:
> @andy-spoo I know Brave and Opera are both Chromium based, but have you given straight Chrome try? I was running into the same issue in Firefox and I tried running from Chrome on another machine and [...]

This is a good point. Personally I only ever use chrome to do my updates so it may be something browser related even though chromium is the underlying engine.

bigtreetech, 2024-07-30:
Wait, @andy-spoo did you solve it? You closed the issue. Posting a soln may help others in the future if you did.

andy-spoo, 2024-07-30:
No I did not close it. I'll give chrome a try in Private mode a bit later.

andy-spoo, 2024-07-30:
So I tried Chrome in private (incognito mode) on this laptop and another laptop with the same result. I tried to see if I could upload an older version and it failed in the same way.

bigtreetech, 2024-07-30:
Out of curiosity, you've tried to re-download the latest 1.0.4 which is now 1.0.4.1 right?

andy-spoo, 2024-07-30:
Yes. I tried 1.0.4 previously and downloaded 1.0.4.1 the other day.
I've just tried factory defaulting the Panda Touch, setting the wifi settings and then trying to upload but still cannot upload.

andy-spoo, 2024-07-30:
Is it possible to install the firmware via either the USB port of the docks USB-C port?

bigtreetech, 2024-07-30:
Only via the USB-C port. @dilixi will assist with that.

andy-spoo, 2024-07-30:
Just to confirm I have tried uploading via Chrome both in and out of incognito mode in both Linux and windows.

andy-spoo, 2024-07-30:
> Only via the USB-C port. @dilixi will assist with that.

I look forward to the instructions.

andy-spoo, 2024-07-30:
I have now also tried uploading via my Android phone. Still no good.

dilixi, 2024-07-30:
> I have now also tried uploading via my Android phone. Still no good.

Hello,please have a try with these methods:
1 open a AP as a WiFi from your phone.
2 use a PandaTouch connect to this AP.
3 close the mobile network from your phone.
4 open a brower on your phone.
5 upload the bin file with v1.0.4
@andy-spoo

dilixi, 2024-07-30:
> Only via the USB-C port. @dilixi will assist with that.
Yes I will assist with this. but we need to do something first.thank you.

dilixi, 2024-07-30:
> > I have now also tried uploading via my Android phone. Still no good.
> 
> Hello,please have a try with these methods: 1 open a AP as a WiFi from your phone. 2 use a PandaTouch connect to this AP. 3 close the mobile network from your phone. 4 open a brower on your phone. 5 upload the bin file with v1.0.4.1 @andy-spoo

andy-spoo, 2024-07-31:
I'm using a Poco X3.
I enabled Portable Hotspot which I presume makes it an access point.
Connected to the Poco from the PandaTouch.
Turned Mobile data off
Tried uploading the version of .bin you mentioned and saw the panda on the phones browser say 97% uploaded and then the same error I've been getting.
It probably went to 100% but it's so quick it's hard to tell.

andy-spoo, 2024-07-31:
As a matter of interest, what is the file uploader for on port 8080 ? As in ipaddress:8080
As well as trying to upload to the regular page using Windows 11 I've now also tried Windows 10 just out of curiosity, using both chrome and Microsoft edge.

bigtreetech, 2024-07-31:
That's for the img binary that contains fonts and graphics.

@dilixi Could the user upload it via USB-C?

andy-spoo, 2024-07-31:
I hope so. I'm running out of ideas.  It would be nice if BTT did the same it does with it's upgrade 3D printer mother boards (I have 3 different versions - which work great, so thanks BTT !!), i.e. allow the use of a firmware.bin file to be inserted and loaded on startup. In this case as a fall back option when the OTA does not work.
I'd imagine in this case it means using avrdude ??

bigtreetech, 2024-07-31:
No, ESP uses a different bootloader that requires the file to be present in the local flash mem. We could check the USB for a file, copy it across and then enter the bootloader I suppose but we have just never considered that because this is the very first time that we have ever seen anything like this. I can only assume that somehow the sector of flash that contains the memory partitioning has an error on your device which is very odd. We can program over USB so it is just a matter of @dilixi showing you how.

andy-spoo, 2024-07-31:
It seems weird to me that uploading a firmware from the original version that was on the Panda to 1.0.3 worked perfectly with zero issues, but going from 1.0.3 to another version fails.

dilixi, 2024-07-31:
> I'm using a Poco X3. I enabled Portable Hotspot which I presume makes it an access point. Connected to the Poco from the PandaTouch. Turned Mobile data off Tried uploading the version of .bin you mentioned and saw the panda on the phones browser say 97% uploaded and then the same error I've been getting. It probably went to 100% but it's so quick it's hard to tell. @dilixi

Hi andy-spoo,
Can you send me a message to this email(1284513393@qq.com). I will give you the tools.
By the way,how many times have you tried using a mobile browser?

andy-spoo, 2024-08-01:
> 1284513393@qq.com

A few times, probably 3. Sending email now.

andy-spoo, 2024-08-02:
@dilixi The update tool worked great!
V1.0.4.1 installed and working like a charm.
Thanks guys!
And happy to have the fat Panda graphic as well :-)

TMillesich, 2024-08-02:
> We can program over USB

So the backplane is a fully functional USB port via pogo pins to the Panda Touch? That's i good to know!

bl4ckm0onk3y, 2024-08-03:
> I wonder if there is not something on your home network that is blocking file transfer on port 80. Maybe try to set up a hotspot on your phone, connect your laptop and the PT and then try again.

i had this problem in BETA channel, it was a WI-FI problem, OP should try to turn off 5Ghz channel if he is using an AP/ROUTER that has 2.4/5Ghz "merged". 

Solution to try, disable 5Ghz network on the router/ap, change the wi-fi channel on 2.4Ghz if the first one didn't work. lastly use your phone as AP and upload the file from the phone.

andy-spoo, 2024-08-03:
> i had this problem in BETA channel, it was a WI-FI problem, OP should try to turn off 5Ghz channel if he is using an AP/ROUTER that has 2.4/5Ghz "merged".
>
> Solution to try, disable 5Ghz network on [...]

I did use my phone as a access point as mentioned in the comments above (and as requested) and it was switched to 2.4Ghz only. It was not the issue here. Also, the same normal setup worked perfectly fine when going from the original pre-installed version to 1.0.3 so it doesn't make sense to me that that would be the issue.

bl4ckm0onk3y, 2024-08-03:
> I did use my phone as a access point as mentioned in the comments above (and as requested) and it was switched to 2.4Ghz only. It was not the issue here. Also, the same normal setup worked perfectly [...]

I had the same problems, exactly the same. I did what i said, my phone worked for a second, then i turned off the 5Ghz because i had merged 2.4/5 channels.

bigtreetech, 2024-08-03:
> > We can program over USB
> 
> So the backplane is a fully functional USB port via pogo pins to the Panda Touch? That's i good to know!

Yes, this is the recovery port that can be used in cases like this one where a user is struggling to program via OTP. It is impossible to "brick" a panda touch thanks to this failsafe.

closed 2024-08-03 by bigtreetech

andy-spoo, 2024-08-04:
> Yes, this is the recovery port that can be used in cases like this one where a user is struggling to program via OTP. It is impossible to "brick" a panda touch thanks to this failsafe.

That is really good to know.

Unixzap, 2024-08-11:
I solve this issue getting HTTP ERROR by just using the Browser DuckDuckGo it work the very first time

MarkusRMZ, 2024-08-14:
@dilixi would you mind to send me ther tool too and the instructions? i keep getting the error. tried with different browsers.

thank you
Markus

dilixi, 2024-08-15:
> @dilixi would you mind to send me ther tool too and the instructions? i keep getting the error. tried with different browsers.
> 
> thank you Markus

https://github.com/dilixi/PandaTouch/blob/master/PandaTouch_recovery.rar

MarkusRMZ, 2024-08-15:
@dilixi Thank for that fast answer. Just a stupid question. How do i connect the Panda? USB-USB to the computer, from where i run the tools? 

thank you,
Markus

Unixzap, 2024-08-15:
Why don't you give duckduckgo a try it worked for no problems

dilixi, 2024-08-15:
> @dilixi Thank for that fast answer. Just a stupid question. How do i connect the Panda? USB-USB to the computer, from where i run the tools?
> 
> thank you, Markus
https://github.com/dilixi/PandaTouch/blob/master/PandaTouch_recovery/how_to_use.doc

You can use the Windows operating system, such as Win10, and then follow the instructions in the document.

Moormerlander, 2024-08-26:
> Why don't you give duckduckgo a try it worked for no problems
> […](#)

Doesn't work either

Moormerlander, 2024-08-26:
What an annoying story! Tried Safari, Google Chrome and DuckDuckGo. I have perfect wifi but this bulls**t update doesn't work. I ordered Panda PWR today, but I think I can't update to 1.05 once released.

Unixzap, 2024-08-26:
You could go to the router and set the Bambu IP address to DMZ just for an update then put it back that’s also something that worked as well

dilixi, 2024-08-26:
> What an annoying story! Tried Safari, Google Chrome and DuckDuckGo. I have perfect wifi but this bulls**t update doesn't work. I ordered Panda PWR today, but I think I can't update to 1.05 once released.

@Moormerlander 
Hi Moormerlander,
Don't worry, please have a try with this tool, thank you.

Moormerlander, 2024-08-27:
> @Moormerlander Hi Moormerlander, Don't worry, please have a try with this tool, thank you. https://github.com/dilixi/PandaTouch/blob/master/PandaTouch_recovery.rar

That is a Windows tool but I have Macs only!

Moormerlander, 2024-08-27:
> @Moormerlander Hi Moormerlander, Don't worry, please have a try with this tool, thank you. https://github.com/dilixi/PandaTouch/blob/master/PandaTouch_recovery.rar

Even if I had a Windows computer: How to connect the Panda Touch to the computer?

dilixi, 2024-08-27:
> https://github.com/dilixi/PandaTouch/blob/master/PandaTouch_recovery/how_to_use.doc

https://github.com/dilixi/PandaTouch/blob/master/PandaTouch_recovery/how_to_use.doc
Hi Moormerlander,you can usb a TYPE-C usb to connect Panda Touch to the computer.

dilixi, 2024-08-27:
> https://github.com/dilixi/PandaTouch/blob/master/PandaTouch_recovery/how_to_use.doc
> Hi Moormerlander,you can usb a TYPE-C usb to connect Panda Touch to the computer.

https://github.com/dilixi/PandaTouch/blob/master/PandaTouch_recovery/how_to_use.doc
Hi Moormerlander,you can usb a TYPE-C usb to connect Panda Touch to the computer.
@Moormerlander

Moormerlander, 2024-08-27:
I opened all ports for the Panda Touch IP without any restrictions but the update process can't be processed. Even support from biqu3d.com only advised this PandaTouch_recovery. I am so disappointed from the makers.

Moormerlander, 2024-08-27:
> https://github.com/dilixi/PandaTouch/blob/master/PandaTouch_recovery/how_to_use.doc Hi Moormerlander,you can usb a TYPE-C usb to connect Panda Touch to the computer. @Moormerlander

But it's useless for Mac users and I am not the only one.

Moormerlander, 2024-08-27:
I even tried the update via Starlink!
both, iPhone and Panda Touch are inside the same LAN, but I still get the annoying error message.

What a crap!

dilixi, 2024-08-27:
> I even tried the update via Starlink! both, iPhone and Panda Touch are inside the same LAN, but I still get the annoying error message.
> 
> What a crap!

Have you tried this method?

1 open a AP as a WiFi from your phone.
2 use a PandaTouch connect to this AP.
3 close the mobile network(NOT THIS AP) from your phone.
4 open a brower on your phone.
5 Upload the bin file (V1.0.4.1)
@Moormerlander

Moormerlander, 2024-08-27:
The update process via phone access point, in this case with an iPhone 14 Pro Max doesn't work either.

[image]
[image]

Moormerlander, 2024-08-27:
> Have you tried this method?
>
> 1 open a AP as a WiFi from your phone. 2 use a PandaTouch connect to this AP. 3 close the mobile network(NOT THIS AP) from your phone. 4 open a brower on your phone. 5 [...]

Yes, same result.

dilixi, 2024-08-27:
> Yes, same result.

Please compress the bin file you are using into a zip file and upload it here for me to check.
1284513393@qq.com

Moormerlander, 2024-08-27:
> > > > I even tried the update via Starlink! both, iPhone and Panda Touch are inside the same LAN, but I still get the annoying error message.
> > > > What a crap!
> > > 
> > > 
> > > Have you tried this method?
> > > 1 open a AP as a WiFi from your phone. 2 use a PandaTouch connect to this AP. 3 close the mobile network(NOT THIS AP) from your phone. 4 open a brower on your phone. 5 Upload the bin file (V1.0.4.1) @Moormerlander
> > 
> > 
> > Yes, same result.
> 
> Please compress the bin file you are using into a zip file and upload it here for me to check. Or send to my Email. [1284513393@qq.com](mailto:1284513393@qq.com)

When you advised me to send you the compressed .bin file I found my mistake!
My Mac didn’t download the file but the html code only.

Please accept my apology for bother you with a problem which was none.
I feel ashamed about my behavior and deleted all my useless comments :-(

Warm regards,

Holger

dilixi, 2024-08-27:
> > > > > I even tried the update via Starlink! both, iPhone and Panda Touch are inside the same LAN, but I still get the annoying error message.
> > > > > What a crap!
> > > > 
> > > > 
> > > > Have you tried this method?
> > > > 1 open a AP as a WiFi from your phone. 2 use a PandaTouch connect to this AP. 3 close the mobile network(NOT THIS AP) from your phone. 4 open a brower on your phone. 5 Upload the bin file (V1.0.4.1) @Moormerlander
> > > 
> > > 
> > > Yes, same result.
> > 
> > 
> > Please compress the bin file you are using into a zip file and upload it here for me to check. Or send to my Email. [1284513393@qq.com](mailto:1284513393@qq.com)
> 
> When you advised me to send you the compressed .bin file I found my mistake! My Mac didn’t download the file but the html code only.
> 
> Please accept my apology for bothering you with a problem which was none. I feel ashamed about my behavior and deleted all my useless comments :-(
> 
> Warm regards,
> 
> Holger

Hi Moormerlander, 
Thank you for your response.
we should check the format of bin file when people upload from browser.
I think you gave some people a lot of useful information to avoid they encountering the same problem in the future.
Can you send me the file you are currently using? We should check at the beginning of the update.
@Moormerlander

Moormerlander, 2024-08-27:
> > > > > > I even tried the update via Starlink! both, iPhone and Panda Touch are inside the same LAN, but I still get the annoying error message.
> > > > > > What a crap!
> > > > > 
> > > > > 
> > > > > Have you tried this method?
> > > > > 1 open a AP as a WiFi from your phone. 2 use a PandaTouch connect to this AP. 3 close the mobile network(NOT THIS AP) from your phone. 4 open a brower on your phone. 5 Upload the bin file (V1.0.4.1) @Moormerlander
> > > > 
> > > > 
> > > > Yes, same result.
> > > 
> > > 
> > > Please compress the bin file you are using into a zip file and upload it here for me to check. Or send to my Email. [1284513393@qq.com](mailto:1284513393@qq.com)
> > 
> > 
> > When you advised me to send you the compressed .bin file I found my mistake! My Mac didn’t download the file but the html code only.
> > Please accept my apology for bothering you with a problem which was none. I feel ashamed about my behavior and deleted all my useless comments :-(
> > Warm regards,
> > Holger
> 
> Hi Moormerlander, Thank you for your response. we should check the format of bin file when people upload from browser. I think you gave some people a lot of useful information to avoid they encountering the same problem in the future. Can you send me the file you are currently using?(I mean the wrong file) We should check at the beginning of the update. @Moormerlander

[panda_touch-v1.0.4.1.bin.zip](https://github.com/user-attachments/files/16761878/panda_touch-v1.0.4.1.bin.zip)

[2 more replies]