bigtreetech/PandaTouch #102: Request - Ability to perform action after print finished.

Vantage-DS, 2024-07-31:
Not sure if this is possible.

For my printers, I usually add GCode to the end of a print to retract 3mm, just to prevent oozing and allow the next print to go down clean, without intervention.

On Repetier I can do this universally, so I don't need to change the slicer machine/filament gcode, it will just apply it to whatever printer I want.

Is it possible to have post printing options on The Panda touch? Turn the light off, pull filament back a few mm, or other features people might like?

bigtreetech, 2024-07-31:
While executing single lines of gcode should theoretically be allowed by the API, it is unfortunately blocked when attempts are made to execute them. Only 3mf files can be printed. A workaround could be to allow print queues and then to completely hack the start code in Bambu studio so that the second 3mf file is nothing more than a retraction command and this is the second print in the queue. Printing queues are something that we have planned in the roadmap.

Vantage-DS, 2024-07-31:
Interesting! That could be a way around it.

Alternatively, the Panda can make the extruder move in both directions via the touch screen, is it possible to have a basic retraction movement happen automatically after a print has finished? Or can it not execute a command without user input?

closed 2025-09-29 by xperiments