bigtreetech/PandaTouch #105: This page isn’t working 192.168.8.103 didn’t send any data. ERR_EMPTY_RESPONSE

Jgelwix, 2024-08-04:
Hello - I continue to experience this issue when attempting to update firmware. I have used various 2.4GHz connections and ensured the IP used in the browser matches the Network Screen/IP Address shown on the device, but all to no avail. The device is connected to WiFi and I can ping the device without issue. I am pinging the device from the same computer where the message below was received. Any guidance would be appreciated.

This page isn’t working 192.168.8.103 didn’t send any data. ERR_EMPTY_RESPONSE

MBP ~ % ping 192.168.8.103
PING 192.168.8.103 (192.168.8.103): 56 data bytes
64 bytes from 192.168.8.103: icmp_seq=0 ttl=255 time=48.044 ms
64 bytes from 192.168.8.103: icmp_seq=1 ttl=255 time=66.771 ms
64 bytes from 192.168.8.103: icmp_seq=2 ttl=255 time=87.999 ms
64 bytes from 192.168.8.103: icmp_seq=3 ttl=255 time=107.387 ms
64 bytes from 192.168.8.103: icmp_seq=4 ttl=255 time=4.622 ms
64 bytes from 192.168.8.103: icmp_seq=5 ttl=255 time=5.331 ms

Thanks in advance.

[image]

bigtreetech, 2024-08-04:
Are you sure that your PC and/or network(s) don't have a firewall blocking port 80 on local connections? It may be better to connect the Panda Touch directly to your phone using your personal hotspot and then to push the firmware file to it directly from the phone.

Jgelwix, 2024-08-05:
Good call - That did the trick. Thank you!

closed 2024-09-29 by BTTJUSTIN