bigtreetech/PandaTouch #106: Add PETG-HF as a selectable filament type

Mynymouqe, 2024-08-04:
Hi team - Bambu has added PETG-HF to their filament range. 

You currently can’t select this on the screen manually, and it displays as PETG (not HF).

Highlander-bdj, 2024-10-25:
This is incredibly annoying.  Can it please be fixed?  It seems like a simple thing to add to the next update.

[image]

mytechguyri, 2024-11-01:
To expand on this further, this screen needs to pull its data from the Bambu cloud instance, as to what filaments are available, as once you configure a custom filament, while you can select that custom filament in the Bambu Studio application, you cannot select it here on the Panda Touch.   More importantly, when a user has said custom filament loaded, and then selects "unload" on the panda touch, they are presented with an "unknown filament type" screen, and are required to select one of the hard coded filament types in order to proceed with the unload operation, simply because its a custom filament assigned in Bambu Studio, and so the panda touch doesn't recognize it.   If it pulled this data from the Bambu cloud server, it would recognize the filament.

bigtreetech, 2024-11-04:
@dilixi

closed 2024-11-04 by Mynymouqe