bigtreetech/PandaTouch #107: Cannot load(use external spool) custom fillament

zeemyself, 2024-08-06:
Panda touch not allow to load custom filament even though it shows PETG-CF value
[image]

https://github.com/user-attachments/assets/60b18bc0-0295-4608-a77d-9918322a8f79

In Bambu/Orca slicer
[image]
[image]

BTTJUSTIN, 2024-10-18:
After an internal discussion,the current PandaTouch's hardware resources cannot meet the request of the custom filament ,we
decided not to implement this feature

closed 2024-10-18 by BTTJUSTIN

Duffy987, 2025-01-08:
The IMG 0233.mov dosnt work. Can help somebody please?

zeemyself, 2025-01-09:
@Duffy987 I have same issue as you 3 months ago it doesn't recognize custom filament same as yours

Duffy987, 2025-01-09:
@zeemyself

Thanks for the information.

bigtreetech, 2025-01-09:
This is being worked on at the moment. Looks like we have found a way to do it and it can be released in the next fw release.

looxonline, 2025-01-09:
1.0.7. The patch versions are just for bug fixes. 1.0.7 should also bring skip object.