bigtreetech/PandaTouch #108: Publish format of .img file

gpambrozio, 2024-08-09:
I know there's a .exe to create img from png but I want to create an automation on my home assistant to update the image automatically with the print image when I start printing. If you tell us how to create an img file I'm sure people will have great ideas on how to improve panda touch.

Thanks.

gpambrozio, 2024-08-10:
Never mind. I reversed engineered and wrote a python script to convert any image to a `new_panda.img`:

```
from PIL import Image
import struct

def process_image(input_file, output_file, max_size=(250, 250)):
    # Open the image using PIL
    with Image.open(input_file) as img:
        # Scale the image down to fit within max_size while maintaining aspect ratio
        img.thumbnail(max_size)

        # Ensure the image is in RGBA mode
        img = img.convert('RGBA')
        width, height = img.size

        # Prepare the binary content
        binary_content = bytearray()

        header = bytes.fromhex('DFAD000001F26E65775F70616E64612E706E67000000')
        binary_content.extend(header)

        # Pack the size as a 4-byte unsigned integer in little endian
        size_bytes = struct.pack('<I', width * height * 3 + 4)
        binary_content.extend(size_bytes)

        dimensions = ((height * 4) << 19) + ((width * 4) << 8) + 5
        dimensions_bytes = struct.pack('<I', dimensions)
        binary_content.extend(dimensions_bytes)

        # Process each pixel row by row
        for y in range(height):
            for x in range(width):
                r, g, b, a = img.getpixel((x, y))

                pixel = (a << 16) + ((r >> 3) << 11) + ((g >> 2) << 5) + (b >>3)
                pixel_bytes = struct.pack('<I', pixel)[:3]

                binary_content.extend(pixel_bytes)

        # Write the binary content to the output file
        with open(output_file, 'wb') as f:
            f.write(binary_content)

```

gpambrozio, 2024-08-10:
I also created an AppDaemon app to update the image every time I start a print (assumes you have the [latest ha integration installed](https://github.com/greghesp/ha-bambulab)):

You'll need to add `Pillow` to `python_packages` section of the add on configuration. Make sure to also update app the parameters on the start of the file with the right image id and urls for your home assistant server and panda touch.

```
import appdaemon.plugins.hass.hassapi as hass

import requests
import struct
from io import BytesIO
from PIL import Image

#
# Update Bambu image
#
# Args:
#
class BambuImage(hass.Hass):
    image_entity = "image.bambu_p1s_cover_image"
    ha_url = "http://home.local:8123"
    panda_url = 'http://192.168.86.88:8080/update_add_file'
    max_size = (250, 250)

    def initialize(self):
        self.run_in(self.image_update, 1)
        self.listen_state(self.image_changed, self.image_entity)

    def image_changed(self, entity, attribute, old, new, kwargs):
        self.run_in(self.image_update, 1)

    def image_update(self, cb_args):
        image_url = self.get_entity(self.image_entity).get_state(attribute="entity_picture")

        if image_url:
            # Fetch the image data
            response = requests.get(f"{self.ha_url}{image_url}")
            if response.status_code != 200:
                self.log(f"Image retrieve failed: {response.status_code}")
                return

            image_data = BytesIO(response.content)
            with Image.open(image_data) as img:
                # Scale the image down to fit within max_size while maintaining aspect ratio
                img.thumbnail(self.max_size)

                # Ensure the image is in RGB mode
                img = img.convert('RGBA')
                width, height = img.size

                # Prepare the binary content
                binary_content = bytearray()

                header = bytes.fromhex('DFAD000001F26E65775F70616E64612E706E67000000')
                binary_content.extend(header)

                # Pack the size as a 4-byte unsigned integer in little endian
                size_bytes = struct.pack('<I', width * height * 3 + 4)
                binary_content.extend(size_bytes)

                dimensions = ((height * 4) << 19) + ((width * 4) << 8) + 5
                dimensions_bytes = struct.pack('<I', dimensions)
                binary_content.extend(dimensions_bytes)

                # Process each pixel row by row
                for y in range(height):
                    for x in range(width):
                        r, g, b, a = img.getpixel((x, y))

                        pixel = (a << 16) + ((r >> 3) << 11) + ((g >> 2) << 5) + (b >>3)
                        pixel_bytes = struct.pack('<I', pixel)[:3]

                        # Append the pixel in BGR format
                        binary_content.extend(pixel_bytes)

                headers = {
                    'Content-Type': 'application/octet-stream;charset=UTF-8',
                }
                response = requests.post(self.panda_url, headers=headers, data=binary_content)
                if response.status_code != 200:
                    self.log(f"Panda Response failed: {response.status_code}")
```

Then add this to `apps.yaml`:

```
bambu:
  module: bambu-image
  class: BambuImage
```

rtcmay, 2024-08-18:
not working Could you tell me how to do it in detail? Thanks

gpambrozio, 2024-08-19:
@rtcmay I created a proper app that you can use to install it using hacs: https://github.com/gpambrozio/BambuAppDaemon.
Take a look and feel free to open an issue on that repo if it doesn't work.

rtcmay, 2024-08-20:
> @rtcmay I created a proper app that you can use to install it using hacs: https://github.com/gpambrozio/BambuAppDaemon. Take a look and feel free to open an issue on that repo if it doesn't work.
ill try thank you su much

rtcmay, 2024-08-20:
@gpambrozio i will try thank you so much

dilixi, 2024-11-19:
> I know there's a .exe to create img from png but I want to create an automation on my home assistant to update the image automatically with the print image when I start printing. If you tell us how to create an img file I'm sure people will have great ideas on how to improve panda touch.
> 
> Thanks.

@gpambrozio 

Hi,
there is [online tool](https://pandatouch.us.kg/) to change the panda image, and you can use it on local by this [rep](https://github.com/bigtreetech-beta/PandaTouch_Addtion_IMG).

And please close this issue thanks.

BTTJUSTIN, 2024-11-20:
This bug was fixed in FW version V1.0.6

closed 2024-11-20 by BTTJUSTIN