bigtreetech/PandaTouch #110: Feature Request: Auto screen activation and screen saver

orangepowerguy, 2024-08-22:
Can the screen auto activate when you start a print and shut down again after the print finishes?

I would also like to see an option for an always on screen (while printing) with the addition of a screen saver that shows the remaining status of a print (% complete, time remaining and current layer number of total layers). When you touch the screen, it could return to normal.

Also looking forward to the thumbnails showing instead of Mr Panda.  :)

Thanks!

BTTJUSTIN, 2024-08-23:
Thanks for making this suggestion. We will discuss it internally and let you know if it will be scheduled for a future firmware release

orangepowerguy, 2024-08-23:
I appreciate the consideration!

aatifk, 2024-11-19:
add a vote for this from me too, if this ends up being possible.

raihei, 2025-03-16:
Auto active the screen when a print starts would be really nice and should be very easy to implement.