bigtreetech/PandaTouch #114: Hide Panda PWR interface in Panda Touch v1.0.5

hugomgteixeira, 2024-09-02:
Hello,

I just updated my Panda Touch fw to version 1.0.5 and now have the interface to use Panda PWR.

While I think the integration will be quite useful for the people using both products, it is quite redundant to have it always present if no Panda PWR is connected. In my case I do not intend to acquire the product in the future as I already have home automation controlling the same features, so it would be quite nice if you could make the interface appear only if the Panda PWR is installed, and hide it if not.

Thank you and continue the good work in improving Panda Touch!

ModeratorPS, 2024-09-02:
I would like this Feature too

TechnoSwiss, 2024-09-03:
Ditto.

BTTJUSTIN, 2024-09-03:
Thanks for making this suggestion. We will discuss it internally and let you know if it will be scheduled for a future firmware release.

jlo1975, 2024-09-03:
yes a simple check box in pandatouch should be great

showtechclove1, 2024-09-07:
Same here. I do not plan to use the PWR, I feel that space should be given back to the main screen.

UsedOak, 2024-09-09:
I am requesting this enhancement as well. While i do intend to purchase the PWR unit, I would prefer to have this usable space back if desired.

davids777, 2024-09-28:
Another request.  I don't plan to use the PWR anytime soon

tghabour, 2024-10-06:
I emphatically second this request

BarakRL, 2024-10-19:
I get that you're trying to up-sale another product which is great, but please let us disable the UI.
(can I revert to 1.0.4.1 safely until you do this?)

ModeratorPS, 2024-10-20:
> I get that you're trying to up-sale another product which is great, but please let us disable the UI.
> 
> (can I revert to 1.0.4.1 safely until you do this?)

Yes you can revert back safely. I have done this too

Tumsk, 2024-10-22:
Another request for this.

bigtreetech, 2024-10-22:
Just a quick one to let you know that we hear you. We didn't really put it there for the upsell. It was put there because that space was pretty much unused previously.

We have 1.0.6 in a very advanced state at the moment and it is jam packed with new features. We got thumbs working nicely, we have support for the new temp/humidity I2C sensor included and a list of other stuff. This tweak will likely only make it into 1.0.7 though.

bigtreetech, 2024-10-24:
Just because @showtechclove1 decided not to show some tech love and give a thumbs down to a fairly reasonable message, we have decided to implement this feature in V1.0.6. Never underestimate the power wielded with a single click of the mighty "thumbs down" ;)

Just kidding. I checked in with the engineers after my last message and they already had it mostly implemented and working fairly well so we figured it made sense to bundle it into the beta of V1.0.6 which is now available on the public beta repo. Enjoy guys.

Highlander-bdj, 2024-11-02:
> Just kidding. I checked in with the engineers after my last message and they already had it mostly implemented and working fairly well so we figured it made sense to bundle it into the beta of V1.0.6 which is now available on the public beta repo. Enjoy guys.

Where is that repo located?

Chief-tl, 2024-11-03:
Great news! When will it be released?

bigtreetech, 2024-11-04:
> > Just kidding. I checked in with the engineers after my last message and they already had it mostly implemented and working fairly well so we figured it made sense to bundle it into the beta of V1.0.6 which is now available on the public beta repo. Enjoy guys.
> 
> Where is that repo located?

It is private so I just added you.

BOFH90, 2024-11-04:
> It is private so I just added you.

Could you please also add me to the repo?

tbloomingdale, 2024-11-04:
Is also would like to be adses to the private repo please.

Tumsk, 2024-11-04:
Please could I have access to the Public Beta Repo?

bigtreetech, 2024-11-04:
@tbloomingdale @Tumsk @BOFH90 All are welcome on the condition that you don't post publicly about bugs in the beta. It is a beta release and it is therefore expected to have some bugs until it reaches release stage. We have had people creating public posts complaining about beta bugs which is why we restrict access.

Chief-tl, 2024-11-04:
Hello, that should be more than understandable. I would also like to test the beta. Thank you!

tghabour, 2024-11-04:
I would also like to be added…thank you!

Marus, 2024-11-06:
Would also like to get access to the beta

Temikus, 2024-11-09:
@bigtreetech Would also like access to the beta. Promise not to spam it and happy to provide feedback if you need 👍

BillWSY, 2024-11-11:
@bigtreetech Hello - I humbly request to be added to the beta repo as well. I have two Panda Touch but since Panda PWR can only pair with one Panda Touch, I figured it would be easier if I could just hide the Panda PWR interface on one of the two Touches.

Will certainly provide feedback to the beta version if I can be helpful.

bigtreetech, 2024-11-11:
All new applicants have been added.

jameswyse, 2024-11-12:
@bigtreetech I'd love beta access as well. You'll only get thumbs up from me 😅 👍

aatifk, 2024-11-19:
please add me to the beta as well! I'd love to test out the new features!

gritstub, 2024-11-21:
@bigtreetech - Hi, also requesting access to beta builds.

SergiyZhyrov, 2025-02-02:
FUUUUUCCKKKK after update from 1.0.5 to 1.0.6, to be able to log-in into account with verification code on email, now " Panda PWR" tab is MiSSED!!!!! i can't setup my Panda PWR, return as was!!! An in READ.me there are not mentioned about this changes!!!!!

closed 2025-03-22 by dilixi