bigtreetech/PandaTouch #119: Feature Request - Ability to print from other plates in .3mf file

RoryMusinskas, 2024-09-07:
Is it possible to add the ability to print from any plate sliced in the .3mf file? 

We run a business and we have lots of different products, each having multiple plates sliced in each .3mf. It would be amazing if it could have the same functionality as the X1C where you can select any plate. 

The alternative is that I have to make a .3mf for each product and variant, which would be hundreds of files.

Thanks!

hugomgteixeira, 2024-09-08:
Agree. Moreover, having to prepare each sliced file from plate 1 is super limitative.

BTTJUSTIN, 2024-09-09:
Thanks for making this suggestion. We will discuss it internally and let you know if it will be scheduled for a future firmware release

bigtreetech, 2024-09-09:
> Is it possible to add the ability to print from any plate sliced in the .3mf file?
>
> We run a business and we have lots of different products, each having multiple plates sliced in each .3mf. It [...]

We have been working on it but it is a challenge. We will continue to work on it though.