bigtreetech/PandaTouch #121: Unable to raise/ lower bed or nozzle tempatures

xmastahux, 2024-09-16:
From firmware version 1.0.4-1.0.5, I have been unable to raise or lower nozzle or bed temperatures using the Panda Touch with my P1S. The Panda Touch does not recognize P1S cache on the machine, and a live service HMS service error occurs occasionally during boot up when not viewing the live feed camera.

bigtreetech, 2024-09-17:
Please read through the wiki manual for answers to all of your questions. bttwiki.com

closed 2024-09-17 by bigtreetech

xmastahux, 2024-09-18:
I did and there was no fix for the issues I mentioned. I emailed service001@biqu3d.com as instructed by the wiki and I will see if I can get help from them instead.

looxonline, 2024-09-18:
[image]

I hope you have a sense of humor and a strong meme game ;)

[image]

looxonline, 2024-09-18:
If you still have issues after taking the recommended actions in the wiki then the service team will help you out via email. They don't monitor github for support queries as it is used mainly by product managers to gather improvements for future firmware updates.

xmastahux, 2024-09-19:
Cool. I will just send it back. You guys are unprofessional af. All you had to do was tell me to contact your support. Instead, you try to insult me by showing me your bad wiki that I may have missed some fine print after reading your wiki several times. My bad. Now, you can close the ticket.

xmastahux, 2024-09-19:
You know what, nevermind. I will return it instead. You guys are impossible
to have any type of good customer experience from the product itself to the
customer service you provide. I honestly hope you guys go out of business
on how you conduct yourselves. Delivering devices that do not perform and/
or defective. Then sending memes to customers looking for help!? Never have
I had a more ridiculous time using your products from the Panda Jet, Panda
Revo, and Panda touch. All three were garbage to install and misleading.
Returning every product I can now and leaving them bad reviews to warn
people of your guys salty attitudes. God forbid a Github user reaches out
for help in regards to their BigTreeTech products because their
intelligence will be mocked. Great job!

philchfree, 2024-09-19:
My take about your issue sounds like Cloud Mode (if you're on Bambu Cloud) is not set for the printer.

Here is extract from wiki:

How To Change Working Mode Between The Panda Touch and Printer
Navigate to the printer/group card page.
Find the printer that you would like to enable or disable cloud connectivity for.
Scroll down to the Edit button for that printer.
Select Enable cloud to set that printer to clould connectivity mode.
TIP You must have a cloud account entered into the settings menu otherwise you will not be able to set any printers to cloud mode.

[image]

looxonline, 2024-09-19:
> I did and there was no fix for the issues I mentioned. I emailed [service001@biqu3d.com](mailto:service001@biqu3d.com) as instructed by the wiki and I will see if I can get help from them instead.

Hey. I saw that you deleted your messages but I still just came to say that I really didn't mean any offense. I know it is difficult to judge tone based on text alone but my message was entirely light hearted and with the intention of pointing you in the right direction. Peace out.

xmastahux, 2024-09-21:
@looxonline That was my bad, please do not take offense to what I said. I read the message from email and thought it was BTT sending me memes in response to the issue I was having. I found that distasteful at first but then realized it was a community member trying to help. It was a bone head mistake on my part and I did laugh at the meme once I realized who sent the message. Terribly sorry for getting upset but I do appreciate your message and understanding. You and @philchfree helped solve my issue, so thank you! I had to go from firmware 1.0.05 > 1.0.03 > 1.0.05 and then run through the steps and wiki outlined. After connecting to the cloud that time, it worked! Thank you for your help again!