bigtreetech/PandaTouch #145: Hidden SSID - can no longer connect in 1.0.5

Candurin, 2024-09-26:
Launch day Panda Touch owner… was able to initially set up my wifi on the panda touch and then re-hide my network SSID.  Worked on 1.0.0, 1.0.1 and 1.0.3.

I can no longer connect to my hidden SSID in 1.0.5.

Any chance you can revert that feature so it works again?

BTTJUSTIN, 2024-09-29:
Thanks for reaching out. Our support team does not operate using the GitHub platform but we would be happy to help you via email based ticketing system. Please visit https://biqu.equipment/pages/technical-support 
and submit a ticket and we will be in touch with you soon.

bigtreetech, 2024-09-29:
@BTTJUSTIN This is a feature request. The user would like hidden SSIDs to be supported.

dilixi, 2024-10-07:
> Launch day Panda Touch owner… was able to initially set up my wifi on the panda touch and then re-hide my network SSID. Worked on 1.0.0, 1.0.1 and 1.0.3.
> 
> I can no longer connect to my hidden SSID in 1.0.5.
> 
> Any chance you can revert that feature so it works again?

Hi ,
Are you sure Pandatouch can connect the hidden ssid in V1.0.3?
I have been check it out ,PandaTouch doesn't support a hidden ssid in currently version.
If you need this feature , I can send you a special bin file.
thanks.

dilixi, 2024-10-07:
> > Launch day Panda Touch owner… was able to initially set up my wifi on the panda touch and then re-hide my network SSID. Worked on 1.0.0, 1.0.1 and 1.0.3.
> > I can no longer connect to my hidden SSID in 1.0.5.
> > Any chance you can revert that feature so it works again?
> 
> Hi , Are you sure Pandatouch can connect the hidden ssid in V1.0.3? I have been check it out ,PandaTouch doesn't support a hidden ssid in currently version. If you need this feature , I can send you a special bin file. thanks(1284513393@qq.com).

Candurin, 2024-10-07:
It did work on all previous firmware versions for me.  It would be great if you could send me the bin file.  Thanks!!

dilixi, 2024-10-07:
> It did work on all previous firmware versions for me. It would be great if you could send me the bin file. Thanks!!

The scan config is same as V1.0.3, there is nothing change.
I'm not sure new config is working well.
Please send me a email.

Candurin, 2024-10-07:
I was able to unhide my network, connect the Panda Touch and then re-hide my SSID.  It would always connect without issue.

This no longer works since updating to 1.0.5.

dilixi, 2024-10-07:
> I was able to unhide my network, connect the Panda Touch and then re-hide my SSID. It would always connect without issue.
> 
> This no longer works since updating to 1.0.5.

Did you have a try with V1.0.3 again?
Does it work?

Vantage-DS, 2024-10-24:
> I was able to unhide my network, connect the Panda Touch and then re-hide my SSID. It would always connect without issue.
> 
> This no longer works since updating to 1.0.5.

This is true. I've since been unable to hide my SSID after 1.0.5.
In previous versions, I un-hid teh SSID, connected, then hid it again.... That doesn't work anymore.

dilixi, 2024-10-24:
> > I was able to unhide my network, connect the Panda Touch and then re-hide my SSID. It would always connect without issue.
> > This no longer works since updating to 1.0.5.
> 
> This is true. I've since been unable to hide my SSID after 1.0.5. In previous versions, I un-hid the SSID, connected, then hid it again.... That doesn't work anymore.

Okay,i will test it later.

dilixi, 2024-10-25:
> This is true. I've since been unable to hide my SSID after 1.0.5. In previous versions, I un-hid the SSID, connected, then hid it again.... That doesn't work anymore.

Hello,
I've tested it and it connects to my hidden Wifi as well.
Which version can works well?

bigtreetech, 2024-10-25:
> Hello, I've tested it and it connects to my hidden Wifi as well. Which version can works well?

Explanation:

V1.0.3 --> User can make SSID not hidden. Connect to SSID. Hide SSID. Panda Touch remains connected even though SSID is hidden.

> V1.0.3 --> User can make SSID not hidden. Connect to SSID. Hide SSID. Panda Touch does not remain connected to SSID after it is hidden.

Could be related to the introduction of the re-scan feature for improved WiFi reliability.

dilixi, 2024-10-25:
> Explanation:
>
> V1.0.3 --> User can make SSID not hidden. Connect to SSID. Hide SSID. Panda Touch remains connected even though SSID is hidden.
>
>
> Could be related to the introduction of the re-scan [...]

Thank you.
 I've tested it by turning off the hidden SSID, it can connect to wifi, and then turning on the hidden , it also can connect to wifi.

Candurin, 2024-11-05:
Interestingly enough.  I noticed that my panda touch screen would never auto-sleep.  I rebooted it from the menu and all of sudden, it now connects to my hidden SSID!

dilixi, 2024-11-06:
> Interestingly enough. I noticed that my panda touch screen would never auto-sleep. I rebooted it from the menu and all of sudden, it now connects to my hidden SSID!

what is the current selection for 'Auto-sleep'? always on?

Candurin, 2024-11-06:
It was always at 15 minutes.  It’s now working properly, as is hidden SSID connecting.

all good!

closed 2024-11-06 by Candurin