bigtreetech/PandaTouch #146: Panda Touch V 1.05

kicksnj1021, 2024-09-26:
Im trying to update the Panda touch units I have to 1.05  I can get the .img file to update fine, but every time I then try and load the .bin file it keeps saying "Invalid File"  Please advise as now I have the Panda's stuck now at the  "OTA is not finished yet" screen

bigtreetech, 2024-09-26:
Don't right click on the link and download. Click on it and then download the raw file using the down arrow in the github UI. There should be a gif in the readme of the firmware folder showing the way.

closed 2024-09-26 by bigtreetech

kicksnj1021, 2024-09-26:
Tried exactly what you said in this email. It still says "File invalid" I
even took a video of the steps taken but it took big to upload. If you
would like the video to see the error message then I will redo it smaller
file size

bigtreetech, 2024-09-26:
Hmmmm, that is odd. Are you sure that you are using the right image file that corresponds to the firmware that you installed? You didn't install an earlier version by mistake?

kicksnj1021, 2024-09-26:
Yes I did.  I even tried to do it on my phone (Android) and still got the
same error. Is there any way to completely factory reset not using the
screen since I can get past the "OTA is not finished yet" screen

bigtreetech, 2024-09-26:
You can use the recovery tool here to update it via USB. https://github.com/bigtreetech/PandaTouch/tree/master/Recovery_tool